JuulLabs/kable #297: Question implementing blocking API calls to characteristic

L3K0V, 2022-04-03:
Hello, I'm playing around with this awesome library, but I think I have get lost somewhere in all scopes and flows.

I checked some of the PRs and Issues and I found that there was some suggestion how to implement `write-for-response` mechanism which I need for a request-response API to be implemented:

```kotlin
private val observation = peripheral.observe(Device.NOTIFY_CHARACTERISTIC)
        .shareIn(scope, started = Eagerly)

private suspend fun writeForResponse(request: ByteArray): ByteArray {
    return observation
        .onSubscription {
            peripheral.write(Device.WRITE_CHARACTERISTIC, request, WriteType.WithResponse)
        }.first()
}

suspend fun init() {
    val data = CommandEnvelope.fromLockVersion(COMMAND_A, device.lockVersion, DEFAULT_AES_KEY)
        .buildCommandPayload()

    println(Hex.encode(writeForResponse(data)))
}

suspend fun getAes() {
    val aesData = CommandEnvelope.fromLockVersion(COMMAND_B, device.lockVersion, DEFAULT_AES_KEY)
        .buildCommandPayload()
    println(Hex.encode(writeForResponse(aesData)))
}
```

And how I used it:

```kotlin
fun main(): Unit = runBlocking {
    println("Start scanning")

    val scanner = Scanner {
        logging {
            filters = listOf(
                Filter.Service(UUID_1),
                Filter.Service(UUID_2)
            )
            level = Logging.Level.Data
        }
    }

    val advertisement = scanner.advertisements.first()

    advertisement?.let { it
        val peripheral = this.peripheral(it.advertisement) {

            logging {
                level = Logging.Level.Data
                format = Logging.Format.Multiline
                data = Hex
            }
        }
        val manager = Manager(it, peripheral, this) // this is the scope

        suspend fun connect() {
            peripheral.connect()
        }

        peripheral.state.onEach { state ->
            if (state is State.Disconnected) {
                connect()
                delay(5_000L)
            } else if (state is State.Connected) {
                manager.init()
                manager.getAes()
            }

        }.launchIn(this)
    }
}

```

What I get is two writes, but only one read on the observation. @twyatt can you give me some hints, suggestions or something. I'm testing it first on my macOS target. Maybe is related with https://github.com/JuulLabs/kable/issues/157

Here's the 🪵:

twyatt, 2022-04-03:
Your implementation looks reasonable to me.

The fact that you're performing writes using `WithResponse` means you _shouldn't_ be affected by #157.

At the tail end of your logs (after the first success write-for-response) you have another `didWriteValueForCharacteristic` callback log, which indicates that Core Bluetooth successfully wrote the data (which is good).
But then 12 seconds elapse and there isn't a callback log of `didUpdateValueForCharacteristic`, which indicates that you never received a response from the remote BLE peripheral.

It is unlikely that incorrect Coroutine scoping would mess with the callback logging (as it is all performed within the Core Bluetooth threads/dispatches and is outside of the Coroutine world — at least at the state of interaction with the underlying Core Bluetooth).

So, the fact that the 2nd `didUpdateValueForCharacteristic` callback isn't logged strongly hints that the peripheral just never responded.

Are you able to test (and confirm) that you see the same behavior on Android?
Happy to investigate further if you are confident that your BLE peripheral is responding and that Kable just isn't picking it up correctly.

L3K0V, 2022-04-04:
@twyatt the mystery is uncovered!

I was trying to implement request-response with larger packages than the MTU, so I started read and write on batches matching the MTU. Everything looks to work like a charm now!

```kotlin
private suspend fun writeForResponse(request: ByteArray): ByteArray {
        return observation.onSubscription {
            request.asSequence().chunked(MTU).map { it.toByteArray() }.forEach {
                peripheral.write(
                    DEVICE.WRITE_CHARACTERISTIC,
                    it,
                    WriteType.WithResponse
                )
            }
        }.transformWhile { bytes ->
            emit(bytes)
            val size = bytes.size
            // check if we reached the package end
        }.fold(byteArrayOf()) { initial, value ->
            val concat = initial.plus(value)
            concat
        }
}
```

Awesome library so far! 🙇‍♂️

closed 2022-04-04 by L3K0V

twyatt, 2022-04-04:
Great to hear that you found the problem!

I'm not sure if Kable could reliably check for it, but if so, it would make sense for Kable to throw an exception if a `write` was attempted that exceeded the MTU.

Created #298 to track the possibility of adding that safety check.