JuulLabs/kable #298: Throw exception if `write` is performed with data size exceeding MTU

twyatt, 2022-04-04:
Assuming that MTU can be reliably checked on all platforms, Kable should throw an exception if a `write` is performed with a payload that exceeds the current MTU.

Currently (at least on Android and Core Bluetooth) it appears that the data is silently not sent, which can cause hard to debug bugs (see #297).

ivanalek, 2023-10-13:
How can we query the MTU for iOS so we don't even perform `write` with a payload that exceeds the current MTU. I can see that [this](https://developer.apple.com/documentation/corebluetooth/cbperipheral/1620312-maximumwritevaluelength) method returns the currently negotiated MTU for iOS, but currently it is not exposed throughout the kotlin code. Is there a plan to improve this in the future or maybe there is another recommended way to do this already?

twyatt, 2023-10-13:
Thanks for the link, I hadn't discovered that method. That'll be helpful when we try to tackle this.
It might be a while before we can prioritize this effort though.
PRs or [discussions](https://github.com/JuulLabs/kable/discussions) (to go over possible solutions/API designs) are welcome in the mean time.