JuulLabs/kable #308: Kable Hangs IOS app when event occurs in background

burnhamd, 2022-04-08:
Occasionally while using bluetooth background mode I see that when I return to the foreground, if an event occurred in the background that the app is hung.  Looking at the trace, I see that it is almost always hung up in the following code:
dispatch_sync(dispatchQueue) {
            block.run()
        }

I can recreate this issue by disconnecting a device multiple times in the background and seeing that it is called from the cancelPeripheralConnection() in connectAsync()

burnhamd, 2022-04-11:
Looking more into this, it is not limited to background.
What I have found is that it fails if there is some action taking place like write, or discovery of services.  

It seems that this is being executed on the ApplePeripheral.connection context.  But the actual write gets executed on the central manager dispatcher context.  The problem seems to be when the connection has been closed but the central manager function is running.  Then it just hangs.  Haven't found a fix.  Im not familiar enough with this code to be able to correct it at this time.

twyatt, 2022-04-11:
Thanks for investigating! Your detailed explanation of the cause is very helpful. When I have some time, I'll explore possible solutions. 👍

closed 2023-08-10 by Phoenix7351