JuulLabs/kable #326: Cancelling a read operation

joharei, 2022-04-26:
I'm not sure if this is caused by Kable or the Android platform. If I cancel a coroutine while it's running a read operation, any attempt to read afterwards immediately fails with a `GattRequestRejectedException` (`message` is `null`). This lasts until I restart the connection. I haven't dared to test the same thing with a write operation 😅

Is this the expected behavior, or a bug in Kable?

My use case is I am reading a characteristic periodically while a specific page is showing, and I want to cancel that polling when navigating away from that page, while keeping the connection open for the next page.

twyatt, 2022-04-26:
Likely a bug in Kable; what you're describing (cancelling a `read` operation) _should_ be supported, and subsequent `read`s _should_ work.

Can you provide logs leading up to the exception?

joharei, 2022-04-27:
Here are logs for two different cases. I used logging `level = Events`. Tell me if I can provide more details.

In this case I kick off a read of characteristic `5fd01`, cancel it, and then try to read `5fcdf`. Reading the second characteristic immediately throws an exception, and then the response from the first characteristic arrives a little later. Notice that `5fd01` takes about 5 seconds to get a response.
```
2022-04-27 09:42:42.257 14877-14877 Kable/Peripheral         D  AC:67:B2:6C:59:E6 read
                                                                service: 000000ff-0000-1000-8000-00805f9b34fb
                                                                characteristic: 10492c5a-deec-4577-a25a-6950c0b5fd01
2022-04-27 09:42:45.625 14877-14877 Kable/Peripheral         D  AC:67:B2:6C:59:E6 read
                                                                service: 000000ff-0000-1000-8000-00805f9b34fb
                                                                characteristic: 10492c5a-deec-4577-a25a-6950c0b5fcdf
2022-04-27 09:42:45.667 14877-14877 Resource                 E  A resource failed with a throwable: com.juul.kable.GattRequestRejectedException
                                                                com.juul.kable.GattRequestRejectedException
                                                                	at com.juul.kable.AndroidPeripheral$read$$inlined$execute$1.invokeSuspend(Connection.kt:106)
                                                                	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
                                                                	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
                                                                	at android.os.Handler.handleCallback(Handler.java:938)
                                                                	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                	at android.os.Looper.loopOnce(Looper.java:201)
                                                                	at android.os.Looper.loop(Looper.java:288)
                                                                	at android.os.HandlerThread.run(HandlerThread.java:67)
2022-04-27 09:42:46.713 14877-15186 Kable/Callback           D  AC:67:B2:6C:59:E6 onCharacteristicRead
                                                                service: 000000ff-0000-1000-8000-00805f9b34fb
                                                                characteristic: 10492c5a-deec-4577-a25a-6950c0b5fd01
                                                                status: GATT_SUCCESS(0)
```

In this case I start reading the same characteristic `5fd01`, cancel it, and wait for the response to arrive. Then I try to read `5fcdf`. At this point I should explain an implementation detail: our device splits data read into chunks of 22 bytes, and we read the same characteristic several times to get the complete data. `5fcdf` is expected to return only 1 byte, but in these logs it returns 22 bytes, tries to read again, and throws. Then the expected 1 byte arrives. Could this be caused by some overflow from the callback of the characteristic that was canceled? I also can't rule out bugs on the bluetooth device 😅
```
2022-04-27 10:16:33.479 16919-16919 Kable/Peripheral         D  AC:67:B2:6C:59:E6 read
                                                                service: 000000ff-0000-1000-8000-00805f9b34fb
                                                                characteristic: 10492c5a-deec-4577-a25a-6950c0b5fd01
2022-04-27 10:16:37.962 16919-17386 Kable/Callback           D  AC:67:B2:6C:59:E6 onCharacteristicRead
                                                                service: 000000ff-0000-1000-8000-00805f9b34fb
                                                                characteristic: 10492c5a-deec-4577-a25a-6950c0b5fd01
                                                                status: GATT_SUCCESS(0)
                                                                data: 00 0A E9 01 0F 41 6E 64 72 6F 69 64 41 50 2D 4A 6F 68 61 6E D3 01
2022-04-27 10:16:45.097 16919-16919 Kable/Peripheral         D  AC:67:B2:6C:59:E6 read
                                                                service: 000000ff-0000-1000-8000-00805f9b34fb
                                                                characteristic: 10492c5a-deec-4577-a25a-6950c0b5fcdf
2022-04-27 10:16:45.118 16919-16919 BluetoothScope           D  Reading characteristic LedState: Read 22 bytes, reading more...
2022-04-27 10:16:45.118 16919-16919 Kable/Peripheral         D  AC:67:B2:6C:59:E6 read
                                                                service: 000000ff-0000-1000-8000-00805f9b34fb
                                                                characteristic: 10492c5a-deec-4577-a25a-6950c0b5fcdf
2022-04-27 10:16:45.123 16919-16919 Resource                 E  A resource failed with a throwable: com.juul.kable.GattRequestRejectedException
                                                                com.juul.kable.GattRequestRejectedException
                                                                	at com.juul.kable.AndroidPeripheral$read$$inlined$execute$1.invokeSuspend(Connection.kt:106)
[... cut ...]
                                                                	at android.os.HandlerThread.run(HandlerThread.java:67)
2022-04-27 10:16:45.188 16919-17386 Kable/Callback           D  AC:67:B2:6C:59:E6 onCharacteristicRead
                                                                service: 000000ff-0000-1000-8000-00805f9b34fb
                                                                characteristic: 10492c5a-deec-4577-a25a-6950c0b5fcdf
                                                                status: GATT_SUCCESS(0)
                                                                data: 02
```

twyatt, 2022-04-27:
Based on the logs you provided and description of what actions you're doing (thanks for the thorough explanation of the issue, btw), I'm fairly confident that `Connections.execute` is not properly handling cancellation.

The gritty details are that normally the lock is held to prevent concurrent Android BLE I/O operations (and to ensure an operation fully completes before attempting a subsequent operation):

https://github.com/JuulLabs/kable/blob/2e013c0541a20376eac913506c3a8a8cb6231879/core/src/androidMain/kotlin/Connection.kt#L54-L73

In your case, cancelling the I/O operation is breaking out of the lock (which is normal and correct coroutine behavior), but then it is allowing a subsequent I/O operation to occur (line 56) before the previous has finished.

I'll try to figure out a fix for this soon. Thanks again for reporting!

joharei, 2022-04-27:
Makes sense. Thanks for the quick response!

twyatt, 2022-09-01:
@joharei I tried to reproduce this issue but was getting `OutOfOrderGattCallbackException` (instead of the `GattRequestRejectedException` you reported), so I'm not sure if I'm actually fixing the issue you were hitting.

I may have a fix, if you can give it a try, please? 🤞 

```kotlin
repositories {
    maven("https://oss.sonatype.org/content/repositories/snapshots")
}

dependencies {
    implementation("com.juul.kable:core:0.18.0-issue-326-1-SNAPSHOT")
}
```

joharei, 2022-09-01:
Hi @twyatt, thanks for looking into this. I've done some initial testing, and it looks like my issue is solved with this fix! I will need to get hold of some devices to verify that my original issue has been solved, but this looks promising.

Thank you!

PR #376 (2023-01-07): Discard response after GATT operation cancellation
Closes #326
closed 2023-01-07 by twyatt