JuulLabs/kable #332: NullPointerException when getting an Android peripheral from advertisement

rsicarelli, 2022-05-27:
After successfully scan and find an advertisement, trying to get the peripheral from it throws a `NullPointerException`:

```kotlin
val peripheral = Scanner()
    .advertisements
    .map { advertisement ->
        //Line below throws NullPointer
        scope.peripheral(advertisement)
    }
    .first()
```
Stacktrace:

```
2022-05-27 10:10:03.018 E/ java.lang.NullPointerException: Parameter specified as non-null is null: method kotlin.jvm.internal.Intrinsics.checkNotNullParameter, parameter <this>
        at com.juul.kable.BluetoothDeviceKt.close(Unknown Source:2)
        at com.juul.kable.AndroidPeripheral$job$1$1.invoke(Peripheral.kt:148)
        at com.juul.kable.AndroidPeripheral$job$1$1.invoke(Peripheral.kt:145)
        at kotlinx.coroutines.JobSupport.invokeOnCompletion(JobSupport.kt:1548)
        at kotlinx.coroutines.JobSupport.invokeOnCompletion(JobSupport.kt:449)
        at com.juul.kable.AndroidPeripheral.<init>(Peripheral.kt:145)
        at com.juul.kable.PeripheralKt.peripheral(Peripheral.kt:99)
        at com.juul.kable.PeripheralKt.peripheral(Peripheral.kt:91)
        at kotlinx.coroutines.flow.FlowKt__TransformKt$onEach$$inlined$unsafeTransform$1$2.emit(Emitters.kt:224)
        at kotlinx.coroutines.flow.FlowKt__ErrorsKt$catchImpl$2.emit(Errors.kt:158)
        at kotlinx.coroutines.flow.FlowKt__ChannelsKt.emitAllImpl$FlowKt__ChannelsKt(Channels.kt:62)
        at kotlinx.coroutines.flow.FlowKt__ChannelsKt.access$emitAllImpl$FlowKt__ChannelsKt(Channels.kt:1)
        at kotlinx.coroutines.flow.FlowKt__ChannelsKt$emitAllImpl$1.invokeSuspend(Unknown Source:14)
        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
        at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
        at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:570)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:749)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:677)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:664)
```

----

### Device configuration:
Name: OnePlus 6
Version: 11

Name: Xiaomi Readmi9T
Version: 11

-------
### More details
- The scan and connection are being made on a [Foreground Service ](https://developer.android.com/guide/components/foreground-services)
- Exception is intermittent and happens every other connection.
- The error seems to happen only on Android >= 10. I've performed tests on multiple devices using Android versions 6, 8, 9 and the exception is not thrown at all, not a single time.

twyatt, 2022-06-02:
Thanks for reporting. Will try to investigate this soon.

twyatt, 2022-06-02:
It shouldn't fail with the `NullPointerException`, so I will look into that, but the stacktrace eludes to the fact that maybe your `scope` is cancelling in the middle of the scan process?

rsicarelli, 2022-06-02:
> It shouldn't fail with the `NullPointerException`, so I will look into that, but the stacktrace eludes to the fact that maybe your `scope` is cancelling in the middle of the scan process?

Yeah you might be correct. I had two different scopes for scanning and connection. After unifying them, the crash is gone.

Not sure if this issue is still relevant, feel free to close if wish 👍

twyatt, 2022-06-03:
> Not sure if this issue is still relevant, feel free to close if wish 👍

I'll close out after I look into if there is a way Kable could throw a more informative failure in the situation you hit.
Thanks for reporting, and following up with how you resolved the issue. 👍