JuulLabs/kable #356: I need to access mBytes data from ScanResult during device search

kemalfaust, 2022-07-14:
Hello everyone. 

I'm using device search  Advertisement class has scanResult and inside this one has mBytes data I need this data during scan.

Device advertise some data for are there new data or not.

Does anyone know how to access this data ?

Advertisement.scanResult.mScanRecord.**mBytes**

twyatt, 2022-07-14:
Are you referring to the `getBytes()` function?

https://developer.android.com/reference/android/bluetooth/le/ScanRecord#getBytes()

kemalfaust, 2022-07-15:
Hi @twyatt yes inside ScanResult there is a method for that but scanResult field private and I can't access 
is it possible the add new public method at Advertisement class

public actual class Advertisement(
    private val scanResult: ScanResult,
)

twyatt, 2022-07-15:
That property can be added, but be aware that it will be Android-only, as (from what I can tell) the other platforms don't provide the same data; so, using such a property would prevent you from supporting the other platforms.

kemalfaust, 2022-07-15:
yes only for Android (probably iOS also has same data but not sure),  do you have plan for adding this property ?
Thanks.

twyatt, 2022-07-15:
> yes only for Android

👍 

> probably iOS also has same data

iOS does not provide the raw data (AFAIK). It is provided (similar to the other platforms) as a map:

```swift
optional func centralManager(
    _ central: CBCentralManager,
    didDiscover peripheral: CBPeripheral,
    advertisementData: [String : Any],
    rssi RSSI: NSNumber
)
```

— https://developer.apple.com/documentation/corebluetooth/cbcentralmanagerdelegate/1518937-centralmanager

> do you have plan for adding this property

It should be pretty simple change, _might_ be able to find some time tomorrow to do it (pretty swamped with another project at the moment, so no promises).

PRs also welcome.

closed 2022-07-15 by twyatt