JuulLabs/kable #358: Cancelling scope of observation prevents restarting observation after creating new scope

twyatt, 2022-07-17:
As reported on [Slack](https://kotlinlang.slack.com/archives/CCSFQJD7V/p1657826674343689), observation isn't starting after a previous observation is cancelled (and remote peripheral is powered off and back on). Upon reconnecting, `setCharacteristicNotification` is not called to enable the observation.

Reproduction steps were said to be:
1. Observe a characteristic.
2. Cancel the scope of observation.
3. Turn off the peripheral (hardware) and turn on it.
4. Reconnect with the peripheral.
5. Observe the characteristic again. (edited)

francismariano, 2022-07-19:
[@twyatt](url) 
the `handler.stopObservation(characteristic)` described above calls [setConfigDescriptor(platformCharacteristic, enable = false)](https://github.com/JuulLabs/kable/blob/25069f74f0f9f8ee7b89c9109d19cdc33d4aa584/core/src/androidMain/kotlin/Peripheral.kt#L390) but setConfigDescriptor never is completed.

setConfigDescriptor calls [write(configDescriptor, DISABLE_NOTIFICATION_VALUE)](https://github.com/JuulLabs/kable/blob/25069f74f0f9f8ee7b89c9109d19cdc33d4aa584/core/src/androidMain/kotlin/Peripheral.kt#L435) but never is completed.

the write function calls [connection.execute](https://github.com/JuulLabs/kable/blob/25069f74f0f9f8ee7b89c9109d19cdc33d4aa584/core/src/androidMain/kotlin/Peripheral.kt#L349) I unable to debug but it never is completed.

twyatt, 2022-07-19:
> the write function calls connection.execute I unable to debug but it never is completed.

Thanks for investigating this! This will be super helpful when I find some time to dig into this.

twyatt, 2022-08-03:
Unfortunately I don't have time to reproduce the issue, so I'm shooting blind with this fix.
If someone is able to test and validate if it actually fixes the issue, it would be much appreciated:

```kotlin
repositories {
    maven("https://oss.sonatype.org/content/repositories/snapshots")
}

dependencies {
    implementation("com.juul.kable:core:0.18.0-issue-358-1-SNAPSHOT")
}
```

ebabel, 2022-08-10:
I've been running with and without this snapshot for the last 8 days and it definitely seems to fix both the issues described above. Thanks @twyatt for fixing and @francismariano for defining the issue! 🎉

twyatt, 2022-08-10:
Thanks for testing/validating the snapshot @ebabel! I've taken the PR out of draft.

PR #368 (2022-08-24): Reset observation state on connect if there are no subscribers
Closes #358
closed 2022-08-24 by twyatt