JuulLabs/kable #365: Crashing with GattStatusException status=GATT_ERROR(133) on Android

ToysoftInc, 2022-07-29:
Our app is crashing 80% of the time when trying to connect to the BLE.
This is from starting the app and then trying to connect.

com.juul.kable.GattStatusException: OnCharacteristicRead(characteristic=00002a28-0000-1000-8000-00805f9b34fb, value=0 bytes, status=GATT_ERROR(133))
        at com.juul.kable.AndroidPeripheral.read(Peripheral.kt:710)
        at com.juul.kable.AndroidPeripheral$read$1.invokeSuspend(Unknown Source:15)
        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
        at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
        at kotlinx.coroutines.internal.LimitedDispatcher.run(LimitedDispatcher.kt:42)
        at kotlinx.coroutines.scheduling.TaskImpl.run(Tasks.kt:95)
        at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:570)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:749)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:677)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:664)
    	Suppressed: kotlinx.coroutines.DiagnosticCoroutineContextException: [StandaloneCoroutine{Cancelling}@afe4f6e, Dispatchers.IO]

twyatt, 2022-08-02:
Unfortunately `133` failure covers many possible failure cases on Android, so it is hard to know what exactly is going wrong for you.

We've commonly seen `133` when a peripheral is out-of-range.

If this is occurs when calling `Peripheral.connect` then you should probably catch this exception and handle as your app sees appropriate (e.g. inform the user that connection failed, or attempt to reconnect automatically).

matsch82, 2022-09-11:
I am facing the same issue. This error is constantly happening. Usually it takes 2 to 3 retries until a connection is established. 
I have seen that connection is stablished with "autoConnect" false .e.g. 
https://github.com/JuulLabs/kable/blob/main/core/src/androidMain/kotlin/BluetoothDevice.kt#L84

which may cause this error according to https://medium.com/@martijn.van.welie/making-android-ble-work-part-2-47a3cdaade07

twyatt, 2023-01-05:
> I am facing the same issue. This error is constantly happening. Usually it takes 2 to 3 retries until a connection is established. I have seen that connection is stablished with "autoConnect" false [...]

Thanks for pointing this out @matsch82, I've rolled this issue into #410.

closed 2023-01-05 by twyatt