JuulLabs/kable #366: App crash for Android 12

keerthankrrao, 2022-08-01:
Experiencing app crash at Android 12, but it works fine for < Android OS v12. I've added the permissions for Android 12 too i.e, Bluetooth Connect and Bluetooth Scan.   

2022-07-28 16:25:36.966 15165-15297/com.skf.tkba.android E/AndroidRuntime: FATAL EXCEPTION: DefaultDispatcher-worker-1
    Process: com.skf.tkba.android, PID: 15165
    com.juul.kable.BluetoothDisabledException: Bluetooth adapter state is Off (10), but On (12) was required.
        at com.juul.kable.PeripheralKt.checkBluetoothAdapterState(Peripheral.kt:499)
        at com.juul.kable.PeripheralKt.access$checkBluetoothAdapterState(Peripheral.kt:1)
        at com.juul.kable.AndroidPeripheral.connect(Peripheral.kt:231)
        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
        at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
        at kotlinx.coroutines.internal.LimitedDispatcher.run(LimitedDispatcher.kt:42)
        at kotlinx.coroutines.scheduling.TaskImpl.run(Tasks.kt:95)
        at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:570)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:749)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:677)
        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:664)
    	Suppressed: kotlinx.coroutines.DiagnosticCoroutineContextException: [StandaloneCoroutine{Cancelling}@e20bf2e, Dispatchers.IO]

twyatt, 2022-08-01:
It appears that the phones bluetooth is turned off.

This is checked via: https://github.com/JuulLabs/kable/blob/fe8ee147df2f282c9defcf1c761296c474991ab5/core/src/androidMain/kotlin/Peripheral.kt#L480-L501

Does this happen on all devices you've tried, or is it specific to certain manufactures or models?
I vaguely remember a specific model would misreport this, and the only workaround was the catch this exception in that situation. 😢

keerthankrrao, 2022-08-05:
Thank you the update. It got worked after adding the permissions for Android 12 and enabling the bluetooth adapter.

twyatt, 2023-01-05:
#335 might help in the future for checking if bluetooth is ready to use.

closed 2023-01-05 by twyatt

twyatt, 2023-12-21:
@degill so this doesn't get lost, created #614 to track.