JuulLabs/kable #375: Unable to retrieve services

lochsenfeld, 2022-08-30:
Currently I am trying to make the SensorTag-Application run on Google Glasses EE2 (Android 8.1.0).
While the app works fine with any other smartphone, I am unable to connect the Glasses to my ble device:

```
2022-08-30 13:36:38.814 3602-3602/my.example.application D/Transponder: connect
2022-08-30 13:36:38.823 3602-3602/my.example.application I/Kable/Peripheral: EC:55:86:5A:2E:3E Connecting
2022-08-30 13:36:38.832 3602-3602/my.example.application D/BluetoothGatt: connect() - device: EC:55:86:5A:2E:3E, auto: false
2022-08-30 13:36:38.832 3602-3602/my.example.application D/BluetoothGatt: registerApp()
2022-08-30 13:36:38.834 3602-3602/my.example.application D/BluetoothGatt: registerApp() - UUID=95fa9451-6609-4512-a53a-b002f18cb3d4
2022-08-30 13:36:38.838 3602-3617/my.example.application D/BluetoothGatt: onClientRegistered() - status=0 clientIf=5
2022-08-30 13:37:19.768 3602-3617/my.example.application D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=5 device=EC:55:86:5A:2E:3E
2022-08-30 13:37:19.770 3602-3617/my.example.application D/BluetoothGatt: onClientConnectionState() - status=34 clientIf=5 device=EC:55:86:5A:2E:3E
2022-08-30 13:37:19.773 3602-3632/my.example.application D/Kable/Callback: EC:55:86:5A:2E:3E onConnectionStateChange
    status: Success
    newState: Connected
2022-08-30 13:37:19.776 3602-3632/my.example.application D/Kable/Callback: EC:55:86:5A:2E:3E onConnectionStateChange
    status: 34
    newState: Disconnected
2022-08-30 13:37:19.776 3602-3632/my.example.application D/BluetoothGatt: close()
2022-08-30 13:37:19.777 3602-3632/my.example.application D/BluetoothGatt: unregisterApp() - mClientIf=5
2022-08-30 13:37:19.777 3602-3602/my.example.application V/Kable/Peripheral: EC:55:86:5A:2E:3E discoverServices
2022-08-30 13:37:19.780 3602-3632/my.example.application D/BluetoothGatt: close()
2022-08-30 13:37:19.781 3602-3632/my.example.application D/BluetoothGatt: unregisterApp() - mClientIf=0
2022-08-30 13:37:19.783 3602-3632/my.example.application D/BluetoothGatt: discoverServices() - device: EC:55:86:5A:2E:3E
2022-08-30 13:37:19.784 3602-3602/my.example.application I/Transponder: Waiting 1000 ms to reconnect...
2022-08-30 13:37:19.785 3602-3602/my.example.application D/BluetoothGatt: close()
2022-08-30 13:37:19.785 3602-3602/my.example.application D/BluetoothGatt: unregisterApp() - mClientIf=0
2022-08-30 13:37:19.788 3602-3602/my.example.application E/Kable/Peripheral: EC:55:86:5A:2E:3E Failed to connect
    
    com.juul.kable.GattRequestRejectedException
        at com.juul.kable.AndroidPeripheral$discoverServices$lambda-12$$inlined$execute$1.invokeSuspend(Connection.kt:106)
        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
        at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
        at android.os.Handler.handleCallback(Handler.java:790)
        at android.os.Handler.dispatchMessage(Handler.java:99)
        at android.os.Looper.loop(Looper.java:164)
        at android.os.HandlerThread.run(HandlerThread.java:65)
2022-08-30 13:37:19.790 3602-3602/my.example.application W/System.err: com.juul.kable.GattRequestRejectedException
2022-08-30 13:37:19.790 3602-3602/my.example.application W/System.err:     at com.juul.kable.AndroidPeripheral$discoverServices$lambda-12$$inlined$execute$1.invokeSuspend(Connection.kt:106)
2022-08-30 13:37:19.790 3602-3602/my.example.application W/System.err:     at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
2022-08-30 13:37:19.791 3602-3602/my.example.application W/System.err:     at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
2022-08-30 13:37:19.791 3602-3602/my.example.application W/System.err:     at android.os.Handler.handleCallback(Handler.java:790)
2022-08-30 13:37:19.791 3602-3602/my.example.application W/System.err:     at android.os.Handler.dispatchMessage(Handler.java:99)
2022-08-30 13:37:19.791 3602-3602/my.example.application W/System.err:     at android.os.Looper.loop(Looper.java:164)
2022-08-30 13:37:19.792 3602-3602/my.example.application W/System.err:     at android.os.HandlerThread.run(HandlerThread.java:65)
```

Approximately 30 seconds elapse between the connect() call and the further logs. My sensor shows that the glasses are connected during the whole period. Then the disconnect occurs with status 34 and a GattRequestRejectedException is thrown.

After my research I find very little on this problem. Maybe a delay between Connect and DiscoverServices has to be done. According to sourcecode, discoverServices should be executed five times. But due to the exception discoverServices is only started once.

Does anyone have a guess as to what the problem might be or if I can set options that will fix it?

During several hours of testing, the sensor was successfully connected once - but had a disconnect after about 30 seconds.

twyatt, 2022-08-30:
> `onClientConnectionState() - status=34 clientIf=5 device=EC:55:86:5A:2E:3E`

According to [here](https://devzone.nordicsemi.com/f/nordic-q-a/50127/error-34-0x22-gatt-conn-lmp-timeout-on-android-devices) and [here](https://e2e.ti.com/support/wireless-connectivity/bluetooth-group/bluetooth/f/bluetooth-forum/677607/cc2640r2f-connection-gets-disconnected), the GATT `34` error is caused by the peripheral not responding to a link parameter update request.

In the 2nd linked page, they said they resolved it by updating the firmware on the peripheral to correctly respond to the request.

Seems like your peripheral isn't configured correctly. You stated you're trying to get the SensorTag sample app the run? If so, I assume you're trying to connect to a SensorTag device; have you updated its firmware?

Perhaps try connecting to another BLE peripheral?

lochsenfeld, 2022-08-31:
Hi @twyatt 
Thanks for the quick response. I am trying to connect to a custom made peripheral.
That is actually a valid point. I will have a chat with the company and my collegues and will give an update as soon as I have more informations.

closed 2023-01-05 by twyatt