JuulLabs/kable #378: Peripheral blocking scope from finishing

tomnovotny7, 2022-09-08:
## Bug report

When using `scope.peripheral(...)` it will block scope from finishing.

I am developing an Android application and I've tried versions `0.18.0`, `0.17.2`, `0.16.1` and unreleased version `0.18.0-issue-359-1-SNAPSHOT` was crashing when using different contexts.

### How to reproduce

A coroutine is not finished unless the job is manually canceled.

```kotlin
val device: BluetoothDevice = ...

viewModelScope.launch {
  val peripheral = peripheral(device)

  // some work, connecting and disconnection or nothing
}
```

Even worse when using a different context:
```kotlin
val device: BluetoothDevice = ...

viewModelScope.launch {
  withContext(Dispatchers.IO) {
    val peripheral = peripheral(device)

    // some work, connecting and disconnection or nothing
  }
  
  // unreachable code
}
```

### Expected behavior

Either should not block scope and be closed automatically or should provide `peripheral.close()` method for closing the peripheral.

twyatt, 2022-09-10:
Thanks for the issue report! It is a good reminder that this portion of Kable is overdue for a redesign.

Short-term: I think providing `peripheral.close()` makes sense.

Long-term: I'd like to remove the reliance on a `CoroutineScope` within a `Peripheral`.

The nitty gritty details are: Initially `CoroutineScope` within a `Peripheral` was used as a crutch, to make passing data across threads simpler/feasible with the old memory model. I believe that with the new memory model, Kable can lean on the threads of the callbacks that are providing the various triggers that need to be acted on (rather than spinning up coroutines). For example, on Apple, we spin up a coroutine to monitor when BLE turns off (to disconnect peripherals) but we should be able to explicitly invoke the appropriate methods from the callbacks themselves (rather than emitting to a flow, to be collected and acted on). In the Android world, a coroutine is spun up to pass along characteristic change events from the current connection (which can change over time); instead of using the `Peripheral`'s `CoroutineScope`, Kable should be using the scope of the collector of characteristic change flow (and it should switch to grabbing events from the current connect, which may change).

twyatt, 2023-09-25:
I'm hoping to have the Android side of this issue resolved in [0.29.0](https://github.com/JuulLabs/kable/milestone/44), the other platforms will be fixed in later versions (I'll keep this issue open until problem is resolved on all platforms).

closed 2024-09-25 by twyatt