JuulLabs/kable #380: Integrate Blue Falcon to support Linux / Windows JVM Targets

danielgruber, 2022-09-19:
Currently, the Kable Library does not support Windows or Linux targets.

There are libraries like https://github.com/weliem/blessed-bluez, that support Windows or Linux. It would be great to investigate if this could be an approach for kable to extend the platform compatibility. I really like the APIs from Kable, so I would prefer using such a library for all platforms instead of using different ones.

saket, 2022-10-12:
Same request for macOS!

twyatt, 2022-10-12:
> Same request for macOS!

@saket MacOS is already supported. 🙂

saket, 2022-10-12:
I'm sorry, I meant support for macOS through JVM

twyatt, 2023-06-21:
I've done some initial investigation into this. DBus seems to be the standard approach to communicating with Bluez on JVM (it is what Blessed Bluez library, that was linked, is doing).

Unfortunately, my development machine is a Macbook, and macOS doesn't have DBus out-of-the-box — it can be set up, but it seemed like a bigger lift than I have time to take on right now.

I might try to get a hold of a Raspberry Pi (with BLE) for testing, but ideally I may look into getting an affordable Linux box for developing this feature.

All that is to say: I am interested in moving the needle on this one, but it will likely be a long ways out.

donadev, 2023-08-07:
Good morning, any update on this? :)
I've been using this library for a while, and I would love to be able to use it in the JVM as well!

twyatt, 2023-08-07:
Unfortunately, no new movement yet. I've been swamped with other projects and haven't yet invested in a machine that I can install Linux on (for developing this feature).

I am very interested in adding support for this, but I imagine it will be a couple months before I can focus on this again. 😢

konpach, 2024-01-21:
Any update here? Could a first step be to provide void implementation(throws e.g. some runtime exception) so it's possible to compile this for jvm along with other targets (iOS, android). Otherwise we need to wrap BLE interface in _expect_ and provide kable as an _actual_ implementation in the available targets.

twyatt, 2024-01-26:
I like that idea, unfortunately I'm swamped with other work projects at the moment.

konpach, 2024-01-26:
I have a PR for you to add this! How to contribute?

konpach, 2024-01-30:
@twyatt if the project is not open for contribution I am happy to send/upload the code change and you can review and add it to the project. I am talking about adding a void jvm implementation as proposed above.

twyatt, 2024-01-30:
Kable is definitely open to contribution, it is under the Apache 2 licence with numerous contributors.

Create a PR and protect maintainers will review it.

konpach, 2024-01-30:
[image]
[image]

@twyatt Creating a PR right away is not possible, only if I fork and propose. Maybe I'm getting something wrong though.

twyatt, 2024-01-30:
Yes, please fork.
You can push your existing branch (`add-jvm-target-with-void-implem`) to your fork, then create a PR.
The head repository will be your fork and branch, and the base repository would be `JuulLabs/kable` with base: `main`.

cedrickcooke, 2024-01-31:
Might be possible to implement with JNI bindings to this: https://github.com/OpenBluetoothToolbox/SimpleBLE

cedrickcooke, 2024-05-16:
Looks like they changed their license so we can't use this without reaching out, but they have exceptions for other open-source BLE libraries like https://github.com/thegecko/webbluetooth, so it might still be possible.

superAndroidMan, 2025-01-20:
anything update?

twyatt, 2025-01-21:
Unfortunately the simpler solutions (e.g. SimpleBLE) are no longer an option (due to licensing restrictions).
So, this will be a fairly large lift (e.g. implement via custom DBus bridge, or similar). I do intend to tackle it one day, but it could be some time before I can dedicate the time required.

vermasrijan19, 2025-02-16:
We have internally started using https://github.com/deviceplug/btleplug with minimal rust wrapper around it and use UniFFI to generate bindings for kotlin.

The code is heavily tangled with our current project but there might be a way for us to contribute, the bindings if maintainers agree with the approach

twyatt, 2025-02-17:
We're definitely open to various approaches. Using Rust bindings might end up being fairly complex and impose a high maintenance burden, but so far, there aren't many other options.

Any chance you could provide a small snippet or example of how these bindings _might_ look in Kable?

vermasrijan19, 2025-03-04:
This is a very rudimentary example I have written for scanning:

Rust code:
`#[uniffi::export(async_runtime = "tokio")]
pub async fn run_ble_scan(filter_string: &str) -> Vec<String> {
    let manager = btleplug::platform::Manager::new().await.unwrap();
    let adapter_list = manager.adapters().await.unwrap();
    let mut device_names = Vec::new();

    for adapter in adapter_list.iter() {
        adapter
            .start_scan(ScanFilter::default())
            .await
            .expect("Can't scan BLE adapter for connected devices...");
        time::sleep(Duration::from_secs(5)).await;
        let peripherals = adapter.peripherals().await.unwrap();
        for peripheral in peripherals.iter() {
            let properties = peripheral.properties().await.unwrap();
            if let Some(local_name) = properties.unwrap().local_name {
                if local_name.contains(filter_string) {
                    device_names.push(local_name);
                }
            }
        }
    }
    device_names
}`

`#[uniffi::export(async_runtime = "tokio")]` ensures that the function is exported and accessible in the .dll file that is generated.

uniffi-bindgen can automatically generate bindings for such exposed functions for multiple languages, kotlin being one of them.

It also generates a lot of boilerplate for type conversion, error handling, loading the dll etc. 

Kotlin code that we have to write:
```
suspend fun main(){
    runBlocking {
        println(runBleScan("Pico"));
    }
}
```

vermasrijan19, 2025-03-04:
https://gist.github.com/vermasrijan19/27ff42bc8355e5d609b16d4305e8ba10 the generated file

twyatt, 2025-03-04:
@vermasrijan19 thanks for the sample! I'll try to find some time to look into this some more.

cedrickcooke, 2025-05-12:
Btleplug + UniFFI turned out to be an excellent approach. I put up a draft PR mostly so I have a place to track what's left to do on that branch, but there's a working proof-of-concept there that's enough to run [JuulLabs/sensortag](https://github.com/JuulLabs/sensortag)

closed 2025-07-21 by cedrickcooke