JuulLabs/kable #382: Can not find a call back for when a write is successful

BryghtLabs, 2022-09-27:
In the code below, anytime we initialize the send characteristic, we don't get a call back if and when the write is successful. This hinders us from using the awesome sdk for firmware updates for our peripherals. Are we missing something, is there a correct way to observe the peripheral for the write characteristic success of failures?

```
suspend fun initSendCharacteristic(service: DiscoveredService) {
        Log.d(TAG, "initSendCharacteristic")
        if (!this::mSendCharacteristic.isInitialized) {
            mSendCharacteristic = service.characteristics.first { it.characteristicUuid == uuidFrom(ChessUpPeripheral.ChessUpSendCharacteristicUUID)}
            Log.d(TAG, "initSendCharacteristic - SEND Characteristic INIT -> $mSendCharacteristic")
            broadcastUpdate(ChessUpBLEService.ACTION_BLE_SEND_CHAR_DISCOVERED)
            MainScope().launch {
                peripheral.observe(mSendCharacteristic).collect { data ->
                    Log.d(TAG, "initSendCharacteristic - Send Characteristic - RECEIVED ${data.size} bytes")
                    Log.d(TAG, "initSendCharacteristic - Send Characteristic - RECEIVED RAW STRING ${String(data)}")
                }
            }
        } else Log.w(TAG, "initSendCharacteristic - SEND Characteristic already initialized")
    }
```

twyatt, 2022-09-27:
I don't quite follow what your issue is. The issue title mentions performing a [`write`], but your code sample doesn't call [`write`] at all. In general, you can [`write`] to a characteristic and it will continue if it was successful or `throw` if it failed, for example:

```kotlin
try {
    peripheral.write(characteristic, byteArrayOf(1, 2, 3))
    // write was successful if we get here
} catch (e: Exception) {
    // write failed if we end up here
}
```

[`write`]: https://juullabs.github.io/kable/core/com.juul.kable/-peripheral/write.html

BryghtLabs, 2022-09-27:
@twyatt Hey thank you for a quick response. Sorry for omitting the "write".  

We will definitely try what you have suggested.

This is what we have done to get around the issue so far:

```
runBlocking {
    peripheral.write(mSendCharacteristic, data, WriteType.WithResponse)
    waitingForOk = true
}
```

closed 2023-01-05 by twyatt