JuulLabs/kable #410: Support `autoConnect` on Android

twyatt, 2023-01-05:
As was mentioned in https://github.com/JuulLabs/kable/issues/365#issuecomment-1243022280, for some users having `autoConnect` hardcoded as `false` may cause connection troubles (specifically, GATT 133 errors).

Kable should provide an option to `connect` with `autoConnect` as `true`.

JonatanPlesko, 2023-06-14:
Do you have any plans for this one?

I'm faced with a different situation where I want to keep a constant connection to a BLE device to observe and display its battery level. I would also like to avoid permanently blocking a thread with connection attempts (until a connection is established) and let the OS connect to the BLE device when available.

twyatt, 2023-06-14:
It is planned, yes, but I'm super busy with work projects, so Kable rarely gets the as much love as I wish I could give it. 🙃 

I try to address Kable development in the following order:

1. features we need internally
2. community submitted PRs
3. issues marked as `bugs`
4. issues with the highest number of 👍 

There is talk of us potentially needing `autoConnect` internally, so this issue may get higher priority when/if that happens.
If you submit a PR I'll make time to help get that in.
Or, slap a 👍 on this issue. 😄

twyatt, 2023-07-03:
Stackoverflow post with good info (for when I'm able to pick this up): https://stackoverflow.com/a/40187086

PR #563 (2023-09-18): Add support for `autoConnect`
[Martijn van Welie](https://medium.com/@martijn.van.welie) wrote a great article, [Making Android BLE work — part 2](https://medium.com/@martijn.van.welie/making-android-ble-work-part-2-47a3cdaade07), which includes a very informative section on [`autoConnect = true`](https://medium.com/@martijn.van.welie/making-android-ble-work-part-2-47a3cdaade07#7fcf).

~~Making `autoConnect = true` the default, as it is generally easier to manage and will match how Kable behaves on Apple platforms.~~

**UPDATE:** Default now matches the previous behavior, library consumers wil [... cut]
closed 2023-09-18 by twyatt