JuulLabs/kable #429: Problem with ATZ command

PavlosTze, 2023-02-15:
Hello, I have the case on Android that we need to run the ATZ command which reboots the device, but no response is coming back from the device (a response like "OK command received I will reboot now"), it just reboots instantly.

I read on your documentation this:

> As before, if the connection drops, the [Flow](https://kotlin.github.io/kotlinx.coroutines/kotlinx-coroutines-core/kotlinx.coroutines.flow/-flow/) will remain active, upon reconnecting the same underlying platform characteristic will be used to to resume streaming characteristic changes.

Here is my code:
```
        if (!write(command)) {
            listener.invoke(Either.Left(BluetoothError.ConnectionRejectedError()))
            return
        }
        
                    readCharacteristic?.let { characteristic ->
                var isSuccess = false
                val inputFlow = peripheral.observe(characteristic)

                inputFlow.takeWhile {
                    val currentResponse = String(it).replace("\r", "").replace("\n", "")
                    if (currentResponse.contains("ERROR")) {
                        Timber.e("[BLE Communication] ERROR: $currentResponse")
                        isSuccess = false
                        return@takeWhile false
                    } else if (currentResponse == "OK") {
                        isSuccess = true
                        return@takeWhile false
                    }
                    true
                }.onCompletion {
                    Timber.d("[BLE Communication]: Success: $isSuccess")
                    if (isSuccess) {
                        listener.invoke(Either.Right(Unit))
                    } else if (it == null) {
                        listener.invoke(Either.Left(BluetoothError.ATCommandError))
                    }
                }.collect()
            }
```

**My problem is** that the listener never invokes, as the BLE Device is getting rebooted and I am waiting for this listener to be invoked in order to proceed further and/or reconnect afterwards.

How to handle it? I tried everything but without the `observe` part it is not working. The BLE Device does not reboots without it.

twyatt, 2023-02-16:
Do you just need to reconnect upon disconnect?
You should be able to handle that within a separate coroutine, this is demonstrated in the SensorTag sample [here](https://github.com/JuulLabs/sensortag/blob/4f1b5cf391eed6e560fab2c22e72e2c127eaca54/app/src/androidMain/kotlin/features/sensor/SensorViewModel.kt#L106-L117).

While that separate coroutine is running, then the `observe` will remain active.
It sounds as though this is not the desired behavior?
If you want the flow to terminate on disconnect, you could use `takeWhile` and have it monitor the peripheral's connection state.

It isn't quite clear to me how you'd like it to behave, so let me know if that doesn't quite answer your question(s).

PavlosTze, 2023-02-16:
I would like the flow to terminate on disconnect, think it as complete reset. I managed to solve it by using the following code:
```
    suspend fun reboot(command: String, listener: (Either<Failure, Unit>) -> Unit) {
        if (!write(command)) {
            listener.invoke(Either.Left(BluetoothError.ConnectionRejectedError()))
            return
        }

        readCharacteristic?.let { characteristic ->
            coroutineScope {
                // TODO: Reboot command doesn't seem to do something.
                val job = launch {
                    peripheral.observe(characteristic).cancellable().collect()
                }
                withContext(coroutineContext) {
                    if (command == AT_REBOOT_COMMAND) {
                        job.cancel()
                        listener.invoke(Either.Right(Unit))
                    }
                }
            }
        }
    }
```

Do you think that's an OK way of handling it?

twyatt, 2023-02-22:
Without having more context about the codebase, here is some general feedback:

1. The `listener` is unnecessary, if you have to invoke the `listener` multiple times, then use a `Flow`; if you only have to return once, you're already a suspend function (so why not just return the result?)
2. I don't understand what the purpose of the `withContext(coroutineContext)` is
3. It isn't clear why you spin up the `job` only to immediately `cancel` it `if (command == AT_REBOOT_COMMAND)`

Maybe you could add more detail around:

> I tried everything but without the `observe` part it is not working. The BLE Device does not reboots without it.

...more specifically, why does the device not reboot w/o the `observe`? Does your firmware need a characteristic observation to be active in order to reboot?

closed 2023-03-10 by twyatt