JuulLabs/kable #433: Bluetooth.availability : to monitor for changes to location services status

francismariano, 2023-02-23:
Here there is a way to monitor for changes to location services status

```
private val locationServiceEnabledFlow: Flow<Boolean> =
    broadcastReceiverFlow(IntentFilter(LocationManager.PROVIDERS_CHANGED_ACTION))
        .map { isLocationEnabled }
```
I think we can combine them to generate a flow with all informations.

francismariano, 2023-02-23:
Hello. I do not know if is the better approach but resolve the issue:

```
private val locationServiceFlow: Flow<Boolean> =
        broadcastReceiverFlow(IntentFilter(LocationManager.PROVIDERS_CHANGED_ACTION))
            .map { isLocationEnabled }
            .onStart { emit(isLocationEnabled) }

    private val bluetoothAvailabilityFlow: Flow<Bluetooth.Availability> =
        broadcastReceiverFlow(IntentFilter(BLUETOOTH_STATE_CHANGED))
            .map { intent -> intent.getIntExtra(EXTRA_STATE, ERROR) }
            .map { state ->
                when (state) {
                    STATE_ON -> Available
                    STATE_OFF -> Unavailable(reason = Off)
                    STATE_TURNING_OFF -> Unavailable(reason = TurningOff)
                    STATE_TURNING_ON -> Unavailable(reason = TurningOn)
                    else -> error("Unexpected bluetooth state: $state")
                }
            }
            .onStart {
                emit(
                    when (BluetoothAdapter.getDefaultAdapter()?.isEnabled) {
                        true -> Available
                        else -> Unavailable(reason = Off)
                    }
                )
            }

    private val bluetoothAvailability: Flow<Bluetooth.Availability> =
        combine(bluetoothAvailabilityFlow, locationServiceFlow) { bluetooth, location ->
            if (SDK_INT <= VERSION_CODES.R && !location) Unavailable(reason = LocationServicesDisabled) else bluetooth
        }
```

Thank you

mmaleiter, 2023-04-15:
Hello everyone,
I really like the kable library and want to offer a solution for this issue. It is more or less the solution from francismariano that I adapted a bit because it felt it is more readable and logical. I am always open for hint and suggestions. I tested it thoroughly and found the solution to be reliable. I would be happy to create a pull request. 

https://github.com/mmaleiter/kable/tree/refine/bluetoothAvailability

twyatt, 2023-04-16:
> P.S. That's my first attempt to contribute to open source, so I hope I did not ignore some obvious conventions.

Hi @mmaleiter. 👋 

You're welcome to create a PR, and we can discuss (via PR comments) any necessary changes needed before it can be merged.