JuulLabs/kable #450: Android: Subsequent reads hang after read fails with GattRequestRejectedException

cleveard, 2023-03-28:
I have run into a problem when a read request fails with GattRequestRejectedException. Once the request fails, then all other read requests do not complete until the device is disconnected and reconnected.

I isolated the issue to miss handling the pending flag added in #376 when action block returns false. Throwing GattRequestRejectedException leaves the pending flag set to true and so subsequent reads hang waiting for a response that is never going to arrive, since the original request failed.

I suppose the same thing would happen if the action block threw an exception, but I am not how that should be interpreted.

This is the relevant code:

https://github.com/JuulLabs/kable/blob/daf79f53677a984d2ed0b9c1c4cddadabb704094/core/src/androidMain/kotlin/Connection.kt#L53-L69

twyatt, 2023-03-28:
Thanks for the bug report! I'll try to find some time soon to look into it.

matejbart, 2023-04-03:
I have stumbled upon the same issue, for android & iOS, and after some debugging I narrowed the issue down to two main causes:

1. The pending flag is always set to true _before_ the action is called. So if the action fails (like in the case how the OP has described it) or if it is cancelled, then the next request will wait for the response of the previous request first, but which will never be received as it hasn't even been sent out.

2. [`Channel.receive()`](https://kotlinlang.org/api/kotlinx.coroutines/kotlinx-coroutines-core/kotlinx.coroutines.channels/-receive-channel/receive.html) may throw a `CancellationException` and sometimes this may be the result of "Undelivered elements" as described by the kotlin doc, i.e. the element in the Channel is lost, but due to the pending flag set to true, the next request will try to receive the previous response first and therefore it will stuck inside the "discard logic" of the previous request the same way as in (1).

I was able to reproduce (1) on android by forcing to throw a `GattRequestRejectedException` in the action and on iOS by setting a long `delay()` inside the action before the read/write and then cancelling the func. I think this can be fixed quite easily by ensuring that the pending flag is only set to true if the action was called successfully.

I could verify (2) only on android by setting a log event for Channel's [`onUndeliveredElement`](https://kotlinlang.org/api/kotlinx.coroutines/kotlinx-coroutines-core/kotlinx.coroutines.channels/-channel.html) callback and then try to aggressively sent a lot of requests in a loop and always cancel some of them. So far, I was only able to fix this issue by "unlocking" the pending flag as soon as the `onUndeliveredElement` has been called.

twyatt, 2023-04-05:
Wups...
`0.22.0-issue-450-1-SNAPSHOT` only contains fix for Android.
`0.22.0-issue-450-2-SNAPSHOT` should have fix for both Android and iOS.

matejbart, 2023-04-06:
> Wups... `0.22.0-issue-450-1-SNAPSHOT` only contains fix for Android. `0.22.0-issue-450-2-SNAPSHOT` should have fix for both Android and iOS.

Thx - I have responded with some feedback directly in the PR.

cleveard, 2023-04-06:
Fixed my issue. I only tested on Android. Thanks!

twyatt, 2023-04-06:
@cleveard, @matejbart had really good feedback on the PR.
Glad it resolves your issue, although it still may suffer "deadlocks" under some conditions.
I'm going to further evaluate possible fixes based on the current feedback on the PR.
Expect additional `SNAPSHOT` builds for testing. Thanks again for helping test!

twyatt, 2023-04-07:
I've published another `SNAPSHOT` with the latest changes on #453:

```
repositories {
    maven("https://oss.sonatype.org/content/repositories/snapshots")
}

dependencies {
    implementation("com.juul.kable:core:0.22.0-issue-450-3-SNAPSHOT")
}
```

@joharei, if you're available to test these `SNAPSHOT`s then that would be greatly appreciated (to ensure these changes don't break the fixes for #326).

cleveard, 2023-04-08:
The new snapshot still fixes my issue.

twyatt, 2023-04-12:
Another snapshot is available:

```
repositories {
    maven("https://oss.sonatype.org/content/repositories/snapshots")
}

dependencies {
    implementation("com.juul.kable:core:0.22.0-issue-450-4-SNAPSHOT")
}
```

I plan to do some final testing of #453, then merging and cutting a release in the next day or two.

joharei, 2023-04-13:
> @joharei, if you're available to test this `SNAPSHOT` then that would be greatly appreciated (to ensure these changes don't break the fixes for #326).

Hey, I did some testing on Android with snapshot 4. No issues 👍

twyatt, 2023-04-13:
> Hey, I did some testing on Android with snapshot 4. No issues 👍

Thank you!

cleveard, 2023-04-13:
Seems OK for me on Android, too.

PR #453 (2023-04-17): Fix BLE I/O cancellation handling
Closes #450
closed 2023-04-17 by twyatt