JuulLabs/kable #458: Logs are duplicated on subsequent reconnects

twyatt, 2023-04-06:
Every next scan  and connect on same device, Kable shows more and more duplicated logs for `DidConnect` and `DidDisconnect`:

```
2023-03-19 12:09:22.563 test.kexe[43203:316424] I/Kable: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 Connecting
2023-03-19 12:09:22.849 test.kexe[43203:316424] D/Kable: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 CentralManagerDelegate state change
  state: DidConnect(identifier=CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2)
2023-03-19 12:09:22.849 test.kexe[43203:316424] D/Kable: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 CentralManagerDelegate state change
  state: DidConnect(identifier=CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2)
2023-03-19 12:09:22.850 test.kexe[43203:316424] D/Kable: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 CentralManagerDelegate state change
  state: DidConnect(identifier=CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2)
2023-03-19 12:09:22.850 test.kexe[43203:316424] V/Kable: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 discoverServices
2023-03-19 12:09:22.897 test.kexe[43203:316443] D/Kable/Delegate: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 didDiscoverServices
2023-03-19 12:09:22.900 test.kexe[43203:316443] D/Kable/Delegate: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 didDiscoverCharacteristicsForService
  service: 6E400001-B5A3-F393-E0A9-E50E24DCCA9E
2023-03-19 12:09:22.900 test.kexe[43203:316424] V/Kable: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 Configuring characteristic observations
2023-03-19 12:09:22.900 test.kexe[43203:316424] I/Kable: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 Connected
2023-03-19 12:09:23.152 test.kexe[43203:316424] D/Kable: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 write
  service: 6e400001-b5a3-f393-e0a9-e50e24dcca9e
  characteristic: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
  writeType: WithoutResponse
2023-03-19 12:09:23.152 test.kexe[43203:316424] D/Kable: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 CentralManager.notify
  service: 6e400001-b5a3-f393-e0a9-e50e24dcca9e
  characteristic: 6e400003-b5a3-f393-e0a9-e50e24dcca9e
2023-03-19 12:09:23.152 test.kexe[43203:316443] D/Kable/Delegate: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 peripheralIsReadyToSendWriteWithoutResponse
2023-03-19 12:09:23.176 test.kexe[43203:316443] D/Kable/Delegate: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 didUpdateNotificationStateForCharacteristic
  service: 6E400001-B5A3-F393-E0A9-E50E24DCCA9E
  characteristic: 6E400003-B5A3-F393-E0A9-E50E24DCCA9E
2023-03-19 12:09:23.299 test.kexe[43203:316443] D/Kable/Delegate: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 didUpdateValueForCharacteristic
  service: 6E400001-B5A3-F393-E0A9-E50E24DCCA9E
  characteristic: 6E400003-B5A3-F393-E0A9-E50E24DCCA9E
2023-03-19 12:09:23.300 test.kexe[43203:316424] D/Kable: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 write
  service: 6e400001-b5a3-f393-e0a9-e50e24dcca9e
  characteristic: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
  writeType: WithoutResponse
2023-03-19 12:09:23.301 test.kexe[43203:316443] D/Kable/Delegate: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 peripheralIsReadyToSendWriteWithoutResponse
2023-03-19 12:09:23.690 test.kexe[43203:316424] I/Kable: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 Disconnected
2023-03-19 12:09:23.690 test.kexe[43203:316424] D/Kable: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 CentralManagerDelegate state change
  state: DidDisconnect(identifier=CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2, error=Error Domain=CBErrorDomain Code=6 "The connection has timed out unexpectedly." UserInfo={NSLocalizedDescription=The connection has timed out unexpectedly.})
2023-03-19 12:09:23.691 test.kexe[43203:316424] D/Kable: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 CentralManagerDelegate state change
  state: DidDisconnect(identifier=CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2, error=Error Domain=CBErrorDomain Code=6 "The connection has timed out unexpectedly." UserInfo={NSLocalizedDescription=The connection has timed out unexpectedly.})
2023-03-19 12:09:23.691 test.kexe[43203:316424] D/Kable: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 CentralManagerDelegate state change
  state: DidDisconnect(identifier=CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2, error=Error Domain=CBErrorDomain Code=6 "The connection has timed out unexpectedly." UserInfo={NSLocalizedDescription=The connection has timed out unexpectedly.})
2023-03-19 12:09:23.691 test.kexe[43203:316424] I/Kable: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 Disconnected
2023-03-19 12:09:24.697 test.kexe[43203:316424] I/Kable/Scanner: Starting scan with native service filtering
```

On third connect the situation is JobCancellationException:

```
2023-03-19 12:11:54.574 test.kexe[43511:318561] I/Kable: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 Disconnected
2023-03-19 12:11:55.580 test.kexe[43511:318561] I/Kable/Scanner: Starting scan with native service filtering
2023-03-19 12:12:06.150 test.kexe[43511:318561] I/Kable/Scanner: Stopping scan
2023-03-19 12:12:06.151 test.kexe[43511:318561] I/Kable: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 Connecting
2023-03-19 12:12:06.242 test.kexe[43511:318561] D/Kable: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 CentralManagerDelegate state change
  state: DidConnect(identifier=CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2)
2023-03-19 12:12:06.242 test.kexe[43511:318561] D/Kable: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 CentralManagerDelegate state change
  state: DidConnect(identifier=CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2)
2023-03-19 12:12:06.242 test.kexe[43511:318561] D/Kable: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 CentralManagerDelegate state change
  state: DidConnect(identifier=CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2)
2023-03-19 12:12:06.242 test.kexe[43511:318561] D/Kable: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 CentralManagerDelegate state change
  state: DidConnect(identifier=CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2)
2023-03-19 12:12:06.243 test.kexe[43511:318561] V/Kable: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 discoverServices
2023-03-19 12:12:06.273 test.kexe[43511:318561] I/Kable: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 Disconnected
2023-03-19 12:12:06.273 test.kexe[43511:318561] D/Kable: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 CentralManagerDelegate state change
  state: DidDisconnect(identifier=CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2, error=Error Domain=CBErrorDomain Code=7 "The specified device has disconnected from us." UserInfo={NSLocalizedDescription=The specified device has disconnected from us.})
2023-03-19 12:12:06.273 test.kexe[43511:318561] D/Kable: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 CentralManagerDelegate state change
  state: DidDisconnect(identifier=CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2, error=Error Domain=CBErrorDomain Code=7 "The specified device has disconnected from us." UserInfo={NSLocalizedDescription=The specified device has disconnected from us.})
2023-03-19 12:12:06.273 test.kexe[43511:318561] D/Kable: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 CentralManagerDelegate state change
  state: DidDisconnect(identifier=CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2, error=Error Domain=CBErrorDomain Code=7 "The specified device has disconnected from us." UserInfo={NSLocalizedDescription=The specified device has disconnected from us.})
2023-03-19 12:12:06.273 test.kexe[43511:318561] D/Kable: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 CentralManagerDelegate state change
  state: DidDisconnect(identifier=CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2, error=Error Domain=CBErrorDomain Code=7 "The specified device has disconnected from us." UserInfo={NSLocalizedDescription=The specified device has disconnected from us.})
2023-03-19 12:12:06.443 test.kexe[43511:318561] E/Kable: CCB61C4A-608F-6662-FE28-CAA5CAAFBDB2 Failed to connect

kotlinx.coroutines.JobCancellationException: LazyDeferredCoroutine was cancelled; job=LazyDeferredCoroutine{Cancelling}@76827b50
```

Is it possible the Peripheral scope to spawn new coroutines in parallel trying to connect in same moment to single device?

_Originally posted by @azlekov in https://github.com/JuulLabs/kable/discussions/445#discussioncomment-5358011_

twyatt, 2023-06-01:
@azlekov if you have some time, can you test if the following `SNAPSHOT` fixes this issue for you?

```kotlin
repositories {
    maven("https://oss.sonatype.org/content/repositories/snapshots")
}

dependencies {
    implementation("com.juul.kable:core:0.23.0-issue-339-1-SNAPSHOT")
}
```

Thank you!

twyatt, 2023-06-08:
This should be fixed in [0.24.0](https://github.com/JuulLabs/kable/releases/tag/0.24.0).

closed 2023-06-08 by twyatt

azlekov, 2023-06-08:
I will try to verify and confirm upcoming weeks! Sorry, busy period...