JuulLabs/kable #474: Testing kable

charlee-dev, 2023-04-30:
Hi. How do we test the kable? Mock Scanner and Advertisements?

twyatt, 2023-05-01:
`Scanner` is an `interface`, so easily mockable; but `Advertisements` should be changed to an `interface`. 👍

charlee-dev, 2023-05-01:
Yes, Scanner itself is not an issue, but Advertisements which is final... if you can abstract it with an interface, that would be great :) Thank you.

twyatt, 2023-05-01:
Ya, I'll try to find some time soon to make `Advertisement` an `interface`. 👍

charlee-dev, 2023-05-01:
From what I see, Android and iOS targets should be pretty straightforward. The Js one will need to change a little bit because of 
```kotlin
public open class AdvertisementImpl( // 1. Changed Advertisement to AdvertisementImpl
    private val advertisement: BluetoothAdvertisingEvent, // 3. 'public' function exposes its 'internal' parameter type BluetoothAdvertisingEvent
): Advertisement { ... } // 2. Extended interface Advertisement
```

twyatt, 2023-05-18:
@charlee-dev does #479 fulfill the use-case you had?

_It will allow mocking of scanner and advertisements, but mocked advertisements won't be usable in `CoroutineScope.peripheral` builder functions._

charlee-dev, 2023-05-18:
I have look. I only use it for scanning. I don't interact with any device.

PR #479 (2023-05-21): Expose `Advertisement` as interface and provide platform specific `Scanner` interfaces
Closes #474.
closed 2023-05-21 by twyatt