JuulLabs/kable #485: Android ANR when collecting advertisements

miwright2, 2023-05-20:
My Android app is in production and uses kable.  It's working great for us.  But I'm getting ANRs from Crashlytics that backtrace to [this line](https://github.com/JuulLabs/kable/blob/2326f416a81fd0af07ac26b45a16276c8013168e/core/src/androidMain/kotlin/Scanner.kt#L49) in kable.

One of the most interesting parts is that the issue effects only Samsung users.  I have ~244 events over 94 users in the last 90 days.

My code essentially looks like this...

```
allMatchesScanner
    .advertisements
    .buffer(capacity = 5, onBufferOverflow = BufferOverflow.DROP_OLDEST)
    .flowOn(Dispatchers.IO)
    .catch { 
        stop()
    }
    .onCompletion {
        Log.d(LOG_TAG, "Scanning stopped ...")
    }
    .collect { advertisement ->
        // process advertising packet
    }
```

And this is the backtrack

miwright2, 2023-05-20:
[inc.combustion.app_issue_4ef85018dda05154ffababd15f0888f2_ANR_session_6466959903A400015B7E1A496343B900_DNE_0_v2_stacktrace.txt](https://github.com/JuulLabs/kable/files/11521293/inc.combustion.app_issue_4ef85018dda05154ffababd15f0888f2_ANR_session_6466959903A400015B7E1A496343B900_DNE_0_v2_stacktrace.txt)

miwright2, 2023-05-20:
[Full trace from Crashlytics](https://github.com/JuulLabs/kable/files/11521298/app_issue_4ef85018dda05154ffababd15f0888f2_ANR_session_6466959903A400015B7E1A496343B900_DNE_0_v2_stacktrace.txt)

twyatt, 2023-05-20:
Internally Kable applies `flowOn(Dispatchers.Main.immediate)` to the `advertisements` flow:

https://github.com/JuulLabs/kable/blob/2326f416a81fd0af07ac26b45a16276c8013168e/core/src/androidMain/kotlin/Scanner.kt#L114

In an experiment, I tried overriding the `flowOn` by doing two consecutive `flowOn`s:

<details>
<summary>Source code</summary>

```kotlin
import kotlinx.coroutines.CoroutineStart.UNDISPATCHED
import kotlinx.coroutines.Dispatchers
import kotlinx.coroutines.GlobalScope
import kotlinx.coroutines.channels.BufferOverflow.DROP_OLDEST
import kotlinx.coroutines.channels.Channel
import kotlinx.coroutines.channels.trySendBlocking
import kotlinx.coroutines.delay
import kotlinx.coroutines.flow.buffer
import kotlinx.coroutines.flow.flowOn
import kotlinx.coroutines.flow.receiveAsFlow
import kotlinx.coroutines.launch
import kotlin.time.Duration.Companion.seconds

suspend fun main() {
    val channel = Channel<Int>()

    GlobalScope.launch(start = UNDISPATCHED) {
        channel
            .receiveAsFlow()
            .flowOn(Dispatchers.Main) // Commenting out this line allows data to "flow".
            .buffer(capacity = 5, onBufferOverflow = DROP_OLDEST)
            .flowOn(Dispatchers.IO)
            .collect {
                delay(1.seconds)
                println(it)
            }
    }

    (0..1_000_000_000).forEach {
        println("--> $it")
        channel.trySendBlocking(it)
    }
}
```
</details>

But it didn't seem to override the `Dispatchers.Main` (as I had expected).

...this is all to say, I don't believe Kable should be opinionated about the dispatcher the flow is on if it can't be overridden.

Please try to following `SNAPSHOT` (which has the `flowOn` removed):

```
repositories {
    maven("https://oss.sonatype.org/content/repositories/snapshots")
}

dependencies {
    implementation("com.juul.kable:core:0.23.0-issue-485-1-SNAPSHOT")
}
```

_It is based on Kable v0.23.0._

mmaleiter, 2023-05-25:
Hello @miwright2, if you still have the issue, I think it is worth to remove the buffer and check if that resolves it.

miwright2, 2023-05-26:
@mmaleiter -- thanks for the suggestion.  

I initially did not have the buffer, but I noticed the `trySendBlocking`.  I'll try removing that buffer with @twyatt changes if the changes alone don't work.  I was also wondering if conflating the flow would help me.  I'm not concerned about occasionally dropping advertising packets because they are connectionless. 

Thanks for the advice and I'll give that approach a shot if needed.

@twyatt -- also, thanks for your quick input!  I'm planning on trying out that snapshot and putting it out in our next release assuming it integrates fine for me.  I'll keep y'all filled in after that release goes live.

twyatt, 2023-05-26:
I think the `buffer` should've been fine to use. I suspect the `flowOn` that Kable used internally was stalling the `trySendBlocking` even before your `buffer` had any affect.

Hopefully w/o the internal `flowOn` (i.e. the `SNAPSHOT`) then the flow will operate as expected (and work w/ or w/o the `buffer`). 🤞 

Keep us posted. Thanks!

nick-sasquatch, 2023-06-23:
Hey @twyatt , I'm working with Matt on this and trying to get the snapshot you mentioned above going, but I'm failing to resolve that snapshot build. Is there a chance it might have been deleted? Thanks!

twyatt, 2023-06-23:
The snapshot index appears to still have it: https://oss.sonatype.org/content/repositories/snapshots/com/juul/kable/core/0.23.0-issue-485-1-SNAPSHOT/

Are you sure you added the snapshot repository?

```kotlin
repositories {
    maven("https://oss.sonatype.org/content/repositories/snapshots")
}
```

nick-sasquatch, 2023-06-23:
> The snapshot index appears to still have it: https://oss.sonatype.org/content/repositories/snapshots/com/juul/kable/core/0.23.0-issue-485-1-SNAPSHOT/
> 
> Are you sure you added the snapshot repository?
> 
> ```kotlin
> repositories {
>     maven("https://oss.sonatype.org/content/repositories/snapshots")
> }
> ```

Yeah, I had--I'm not sure why, but I had to blow away a couple caches, and now it works. Thanks for the quick response!

miwright2, 2023-07-11:
> @twyatt -- Did the changes you made for this issue make it into `0.24.0`?

@twyatt -- nevermind, I see that it isn't in `0.24.0`.

twyatt, 2023-07-11:
@miwright2 I was waiting for a report back that it resolved the issue before cutting a release with that change. As it does have an impact on library consumers, so I wanted to be sure that was the actual cause/fix for the problem.

miwright2, 2023-07-11:
@twyatt -- sure thing -- we are about to release using that snapshot.  I will let you know how it goes.  I just wanted to make sure that it didn't already make it into a release.

miwright2, 2023-07-25:
@twyatt -- this fix looks good for us as far as the ANR.

twyatt, 2023-07-25:
Thanks for verifying the ANR fix!

I'll cut a release shortly after #537 has the necessary approvals and is merged.

As for the `NullPointerException` you're seeing, let's track that in a separate issue.

closed 2023-07-26 by twyatt

twyatt, 2023-07-27:
@miwright2 ANR fix was included in [0.25.0](https://github.com/JuulLabs/kable/releases/tag/0.25.0) release.

PR #537 (2023-07-26): Remove `flowOn(Main.immediate)` from `advertisements`
Closes #485