JuulLabs/kable #497: OutOfOrderGattCallbackException: Unexpected response type OnCharacteristicWrite received

egmazur, 2023-06-07:
I work on Android. Have been using this lib for more than one year. And now I got the case when I catch `OutOfOrderGattCallbackException`. The full text is `OutOfOrderGattCallbackException: Unexpected response type OnCharacteristicWrite received`. The scenario is pretty simple: there are two characteristics. One I use to write byte arrays on a device, the second one is read the status of this "writing". And in one moment I get this exception. I found it is source code but its cause is not clear for me. Why do I get it?

twyatt, 2023-06-12:
What version of Kable are you using?

`OutOfOrderGattCallbackException` occurs when the underlying bluetooth callback provided Kable with an unexpected response.

More specifically, Kable ensures that every I/O operation happens within a lock. For example, when a **write** request is initiated, the lock prevents other I/O operations until Kable gets a **write** response. If an unexpected response is received, i.e. receiving a **read** response callback when we had just performed a **write** request, then `OutOfOrderGattCallbackException` would be thrown.

In most cases, this should not occur. I haven't seen the Android system giving unexpected responses, but that is what that exception is intended to inform you of (that failure case).

There is some complexity around cancellation handling, so it is possible that the internal Kable bookkeeping has a bug.

I would need more detailed logs to know more, and to help further.

grabbe-bryghtlabs, 2024-04-09:
We came upon this issue with version 0.23.0.

twyatt, 2024-04-09:
@grabbe-bryghtlabs can you provide logs to help debug the issue?

grabbe-bryghtlabs, 2024-04-09:
This is a snippet
`2024-04-09 11:24:35.854 28595-28632 oplus_bt_s..._ble_event com.android.bluetooth                I  report_stack_ble_gatt_read_write_operation_event enter
2024-04-09 11:24:35.854 28595-28632 BtGatt.GattService      com.android.bluetooth                D  onWriteCharacteristic() - increasing permit for address=DC:DA:0C:15:1F:A6
2024-04-09 11:24:35.855 28595-28632 oplus_bt_s..._ble_event com.android.bluetooth                I  stack_ble_gatt_read_write_operation_event_report address=dc:**:**:**:1f:a6
2024-04-09 11:24:36.062 28595-28916 BtGatt.GattService      com.android.bluetooth                D  writeCharacteristic() - trying to acquire permit.
2024-04-09 11:24:36.100 28595-28795 bt_bta_gattc            com.android.bluetooth                D  Handling write response type: 2: handle 0x000c
2024-04-09 11:24:36.100 28595-28632 oplus_bt_s..._ble_event com.android.bluetooth                I  report_stack_ble_gatt_read_write_operation_event enter
2024-04-09 11:24:36.101 28595-28632 BtGatt.GattService      com.android.bluetooth                D  onWriteCharacteristic() - increasing permit for address=DC:DA:0C:15:1F:A6
2024-04-09 11:24:36.103 28595-28632 oplus_bt_s..._ble_event com.android.bluetooth                I  stack_ble_gatt_read_write_operation_event_report address=dc:**:**:**:1f:a6
2024-04-09 11:24:38.906  1299-6353  ANDR-PERF-LM            ven...qti.hardware.perf@2.2-service  E  AdaptiveLaunch: writeToDataBase() 64: Meter aborted or could not get meter data for this run
2024-04-09 11:24:41.413  1299-1358  ANDR-PERF-LM            ven...qti.hardware.perf@2.2-service  W  FeatureState: writeState() 188: GameOptimization: State transition not allowed
2024-04-09 11:24:42.440  1299-6353  ANDR-PERF-LM            ven...qti.hardware.perf@2.2-service  E  AdaptiveLaunch: writeToDataBase() 64: Meter aborted or could not get meter data for this run
2024-04-09 11:24:44.800  1299-1358  ANDR-PERF-LM            ven...qti.hardware.perf@2.2-service  W  FeatureState: writeState() 188: GameOptimization: State transition not allowed
2024-04-09 11:24:45.012  1299-1358  ANDR-PERF-LM            ven...qti.hardware.perf@2.2-service  W  FeatureState: writeState() 188: GameOptimization: State transition not allowed
2024-04-09 11:24:46.051  1299-1358  ANDR-PERF-LM            ven...qti.hardware.perf@2.2-service  W  FeatureState: writeState() 188: GameOptimization: State transition not allowed
2024-04-09 11:24:48.988 17668-17668 BLEBoard                usap64                               E  Failed to send data, timeout
                                                                                                    com.juul.kable.GattRequestRejectedException
                                                                                                    	at com.juul.kable.BluetoothDeviceAndroidPeripheral$write$$inlined$execute$1.invokeSuspend(Connection.kt:80)
                                                                                                    	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
                                                                                                    	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:958)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                                                    	at android.os.Looper.loopOnce(Looper.java:257)
                                                                                                    	at android.os.Looper.loop(Looper.java:368)
                                                                                                    	at android.os.HandlerThread.run(HandlerThread.java:67)`

twyatt, 2024-04-09:
@grabbe-bryghtlabs thanks! I'll also need [Kable logs](https://github.com/JuulLabs/kable?tab=readme-ov-file#logging) leading up to the issue. With level `Events` or `Data`.

twyatt, 2024-04-09:
Thanks! The very first line of those logs show a write to a **characteristic** but then shortly after you see the exception from Kable stating that it received a **descriptor** response.

Is there any chance you can provide more of the logs leading up to the crash?
I'm trying to understand what data was sent to Android and what sequence Android is responding in. Thanks again.

grabbe-bryghtlabs, 2024-04-10:
This is after updating to 0.31.0

twyatt, 2024-04-11:
I have a vague suspicion that maybe you have two `Peripheral` objects for the same connection?
Can you look through how you're creating the peripheral and if there is a way you possible have multiple objects (rather than references to the same object).
I'm guessing that is why multiple I/O operations are occurring at once.

If you're having trouble isolating such a situation in your code, let me know, and I can add additional logging that will confirm/deny my suspicion.

grabbe-bryghtlabs, 2024-04-11:
The way it is built out we have a single scan job and one instance of a peripheral.  I don't think that would be the case.  If there is a way that I can provide more logging, please let me know.  I can run through the same scenario in an isolated environment this evening.

twyatt, 2024-04-18:
I have some theories of what might be going wrong. I'll try to add some necessary logging to validate soon. Thanks for being patient.

grabbe-bryghtlabs, 2024-04-18:
Just let me know what I can do.  I appreciate your attentiveness

grabbe-bryghtlabs, 2024-04-23:
I'm also finding this exception cropping up when using WithResponse

djweber, 2024-05-31:
Just wanted to confirm that I've seen this happen to around 5 of our customers.

```  
Fatal Exception: com.juul.kable.OutOfOrderGattCallbackException: Unexpected response type OnCharacteristicWrite received
       at com.juul.kable.BluetoothDeviceAndroidPeripheral.read(BluetoothDeviceAndroidPeripheral.kt:782)
       at com.juul.kable.BluetoothDeviceAndroidPeripheral$read$1.invokeSuspend(:15)
       at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
       at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:104)
       at kotlinx.coroutines.internal.LimitedDispatcher$Worker.run(LimitedDispatcher.kt:111)
       at kotlinx.coroutines.scheduling.TaskImpl.run(Tasks.kt:99)
       at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:584)
       at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:811)
       at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:715)
       at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:702)```
       
Our current setup has peripherals cached in a map by their advertisement identifier (mac address), so there shouldn't be a case where we are using the same peripheral in two different spots.

grabbe-bryghtlabs, 2024-05-31:
> Just wanted to confirm that I've seen this happen to around 6 of our customers. There are several thousand using the latest build, so the volume is relatively low.
> 
> ```
> Fatal Exception: com.juul.kable.OutOfOrderGattCallbackException: Unexpected response type OnCharacteristicWrite received
>        at com.juul.kable.BluetoothDeviceAndroidPeripheral.read(BluetoothDeviceAndroidPeripheral.kt:782)
>        at com.juul.kable.BluetoothDeviceAndroidPeripheral$read$1.invokeSuspend(:15)
>        at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
>        at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:104)
>        at kotlinx.coroutines.internal.LimitedDispatcher$Worker.run(LimitedDispatcher.kt:111)
>        at kotlinx.coroutines.scheduling.TaskImpl.run(Tasks.kt:99)
>        at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:584)
>        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:811)
>        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:715)
>        at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:702)
> ```
> 
> Our current setup has peripherals cached in a map by their advertisement identifier (mac address), so there shouldn't be a case where we are using the same peripheral in two different spots. For what it's worth, our peripherals are generally connected to and disconnected from throughout the day. In either case, I can try to put some defense around our I/O operations. Would it be preferable to just reconnect the peripheral if this happens, or is this a recoverable scenario during a connection?

Not sure about your scenario, but I've had this come up when updating firmware on a BLE device and it gets bricked.  We have another process to force our firmware to update, but it's a bit cumbersome and it would be nice to figure out why it's happening and if I can do anything to guard against it.  We also have the peripheral in memory in a singleton.  For general calls I have things wrapped in a retry with delay.

twyatt, 2024-09-19:
I don't have a reliable way of reproducing this, but #749 should fix this (and it slated to be included in the upcoming [0.35.0](https://github.com/JuulLabs/kable/milestone/53) release).

closed 2024-09-25 by twyatt