JuulLabs/kable #504: Bluetooth Permissions Modal shows prematurely

twyatt, 2023-06-15:
Reported internally:

> I've tracked the issue down to CBCentralManager initialization in Kable in the CentralManager class. It seems that initialization–itself–is enough to trigger the prompt. I deleted all usages of the local cbCentralManager and did not provide a delegate in the init, so it doesn't happen when the manager is used–happens right when it's init'd. What's weird is that this isn't the behavior in iOS. I can initialize a CBManager natively without being prompted.

Phoenix7351, 2023-06-23:
Was wondering what @twyatt and @davidtaylor-juul 's thoughts are on this? Based on my findings, a reasonable solution would be to find a way to defer the initialization, but I thought there might be something I'm missing? Or maybe a better solution?

davidtaylor-juul, 2023-06-23:
I think we should investigate further the discrepancy between natively initializing (and no prompt) vs initializing in this library and having an immediate prompt.

twyatt, 2023-06-24:
@Phoenix7351 it'd be great to get a sample iOS app in the [SensorTag sample app](https://github.com/JuulLabs/sensortag) (it has been [requested](https://github.com/JuulLabs/sensortag/issues/11)); if you have the time/bandwidth?

With iOS support in the SensorTag sample, then it would give me a nice simple repro to be able to easily investigate this issue.

Phoenix7351, 2023-06-24:
> @Phoenix7351 it'd be great to get a sample iOS app in the [SensorTag sample app](https://github.com/JuulLabs/sensortag) (it has been [requested](https://github.com/JuulLabs/sensortag/issues/11)); if [...]

Agreed, but, yeah, really don't have the bandwidth for it at the moment. It might be helpful to checkout my CA-2008 branch? I did a lot in there to stub out things to eliminate them as potential causers of this issue. In that branch you can bring in/comment out the CBCentralManager to replicate the behavior. That might be helpful to start?

Phoenix7351, 2023-06-24:
It just occurred to me that the issue is probably not having the iOS app side setup. I can help you with that!

Phoenix7351, 2023-06-24:
https://medium.com/@adrianwitaszak/kotlin-multiplatform-mobile-bluetooth-permission-and-bluetooth-service-setting-9a697e70700

This article makes no mention of this behavior, but also instantiates `CBCentralManager` lazily.

PR #739 (2024-08-24): Prevent bluetooth permission dialog when spinning up `CBCentralManager`
> [!WARNING]
> This PR is a part of a larger refactor.

Closes #504
closed 2024-08-24 by twyatt