JuulLabs/kable #508: Cast exception when using Peripheral.identifier when extending AndroidPeripheral for testing

rsicarelli, 2023-06-20:
## Context
I am getting a `java.lang.ClassCastException` when using the `AndroidPeripheral` interface for tests:

```kotlin
class FakeAndroidPeripheral(
    override val address: String = "43:09:19:39:99",
    ...
) : AndroidPeripheral

val fakeAndroidPeripheral = FakeAndroidPeripheral()

val identifier = fakeAndroidPeripheral.identifier
```

throws:

```
java.lang.ClassCastException: class FakeAndroidPeripheral cannot be cast to class com.juul.kable.BluetoothDeviceAndroidPeripheral (FakeAndroidPeripheral and com.juul.kable.BluetoothDeviceAndroidPeripheral are in unnamed module of loader 'app')
	at com.juul.kable.PeripheralKt.getIdentifier(Peripheral.kt:467)
```

Do you have any ideas on how to work around this problem?

Thanks a lot!

twyatt, 2023-06-20:
The `identifier` property is an extension function (legacy design before each platform got its own peripheral interface — done in Kable 0.23.0).

I'll drop the extension function and promote `identifier` as a property on the `Peripheral` interface so that it can be overridden (faked).

twyatt, 2023-06-22:
@rsicarelli, #509 should resolve this issue. You can test it (before it is available in the next Kable release) with:

```kotlin
repositories {
    maven("https://oss.sonatype.org/content/repositories/snapshots")
}

dependencies {
    implementation("com.juul.kable:core:0.24.0-issue-508-1-SNAPSHOT")
}
```

rsicarelli, 2023-06-22:
@twyatt Works like a charm, thanks for the quick response and fix!

PR #509 (2023-06-26): Promote `identifier` to property on `Peripheral` interface
Closes #508.
closed 2023-06-26 by twyatt