JuulLabs/kable #517: Exposure identifier in Advertisement just like is exposured in Peripheral object

francismariano, 2023-07-03:
Hello everyone.

What do you think to exposure the identifier (macadress, unique identifier) in Advertisement object? The address already is available in AndroidAdvertisement but it is good to have the identifier available in common code as well.

Use case: mount a list with advertisement founded directly from common code.

Thank you.

twyatt, 2023-07-03:
Having the `Peripheral` identifier and `Advertisement` identifier differ definitely isn't ideal — and I would like to bring them into alignment, I was just torn between which approach was better.

The trade-offs I see are:

With having an identifier in common code is obviously much more convenient to use, the drawback being that in common code you need to use a _common_ type (in this particular case, `String` is a reasonable choice for this, IMHO), but `String` doesn't provide the type safety that the more correct type would offer.

Granted, on Android, the `address` (which backs the identifier) is a MAC address represented as a `String`, but Kable could theoretically (in the future) provide a `MacAddress` type. 🤷 

Currently, the Apple `Advertisement` provides `identifier` as an `NSUUID` which is platform specific. It could easily be converted to `Uuid` to be multiplatform friendly, but it still isn't compatible with the identifiers of the other platforms.

So, my question to libraries consumers is (knowing the trade-offs), which is preferred?

1. Identifiers on platform specific interfaces (with their appropriate types) — _how advertisements are now_
2. Identifier on the root (common) interface as `String` — _how peripherals are now_

twyatt, 2023-07-03:
@francismariano I am leaning towards option number 2 (as this issue is asking for), but want to give it a little time to see if others stumble on this issue and have input/opinions on the matter.

twyatt, 2023-07-03:
Actually, I just thought of an option 3, which is option 2 with the addition of adding a property to the Apple Advertisement (and peripheral) interface which would return the identifier as a `Uuid`. Not sure of a good name for that property, I guess `platformIdentifier` — it isn't a great name, so suggestions are welcome.

twyatt, 2023-07-03:
@francismariano I added a bunch of comments, only to realize that I used `expect`/`actual` with type aliasing for the `identifier` on the `Peripheral`. It does make sense to just carry over that approach to the advertisement. 👍

francismariano, 2023-07-03:
@twyatt I agree to just carry over that approach to the advertisement. By the way, I am trying to do that here and offer um pull request for you, but I am with problem related to atomicfu after publish the kable to maven local. Dunno the reason yet.

PR #518 (2023-07-06): Add identifier property to Advertisement object
Closes #517
closed 2023-07-06 by twyatt