JuulLabs/kable #521: [Feature] Scanner: easy way to detect when device is no longer available

rocketraman, 2023-07-05:
The Scanner should provide a way to determine when a device is no longer available, because it has been turned off or is no longer in range.

As I understand it, at least on Android, the right way to do this is to scan again, and determine from the new scan if any devices are no longer present relative to the previous scan.

See https://stackoverflow.com/questions/33033906/how-to-detect-when-a-ble-device-is-not-in-range-anymore.

twyatt, 2023-07-06:
I'm reluctant to implement such a feature in Kable since I believe that should be considered "no longer available" comes down a lot to the needs of the app that is using Kable.

Additionally, I think implementing such logic app-side should be fairly straight forward?

Something to the effect of (pseudo code):

```kotlin
val pass1 = withTimeout(11.seconds) {
    scanner.advertisements.toList()
}
val pass2 = withTimeout(11.seconds) {
    scanner.advertisements.toList()
}
val noLongerAvailable = pass1 - pass2
```

rocketraman, 2023-07-06:
Thanks I'll experiment with this. If it does work, then what do you think about a utility method on advertisements like `scanOneInterval()`? That would abstract away the low-level detail of the 10.24 seconds that covers one advertising interval.

twyatt, 2023-07-06:
I'm reluctant to add such a utility method only because (as far as I understand) advertisement interval is customizable (on the peripheral side), so some firmware might use a more frequent interval while others use less, and 10.24 is just the maximum — so the naming `scanOneInterval` may not be accurate for all peripherals.

The timeout could be configurable and default to `10.24`, but that becomes a lot of code for something that is fairly simple to implement on the consumer side?

Ultimately, I appreciate the suggestion and always open to further discussion (I'm just cautious about growing the API unnecessarily).

rocketraman, 2023-07-08:
This approach works but there are a few complexities to keep in mind.

1) You can't use `toSet` directly, as (at least for my device) each `AndroidAdvertisement` is not equal, even though its the same address. Mapping to the `address` field of the advertisement is needed for the set comparison.

2) Obvious in hindsight but caused some problems initially: BLE devices stop advertising when they are connected. Therefore, we can't assume an already connected device is "gone" relative to the current device when it is no longer present in the advertisements.

I went with an approach that looks something like this. My peripheral advertises every 250 ms so a 1 second chunk time was sufficient, and this approach does not require multiple separate scans:

```kotlin
      var visibleAddresses = emptySet<String>()
      // any peripherals already connected from other app state
      var connectedAddresses = setOf(...)

      scanner.advertisements
         // custom implementation
         // see https://github.com/Kotlin/kotlinx.coroutines/issues/1302
        .chunked(100, 1.seconds)
        .map { it.associateBy { it.address } }
        .distinctUntilChanged()
        .collect { ads ->
          val removed = lastAddresses - connectedAddresses - ads.keys
          val added = ads.keys - lastAddresses
          if (added.isNotEmpty() || removed.isNotEmpty()) {
            lastAddresses = ads.keys
          }
        }
```

If you are curious about my implementation of `chunked` I can share it.

I don't feel this is trivial to do for an application consumer, so to me it makes sense to have some of this logic in Kable. However, I understand not wanting to expand the API surface. Feel free to close this issue.

twyatt, 2023-07-08:
I see. You've definitely exposed more complexity than I had original anticipated.

I'll keep the issue open, as it allows others to discover the feature request and 👍 if they too would find it useful. That would at least gauge interest.

I'm still on the fence as to if it should be added to Kable, especially with needing a custom (`chunked`) operator, makes for a potentially larger maintenance burden. Lots of tradeoffs to consider. Regardless, I would appreciate you sharing your `chunked` implementation, it would prove useful if later we do decide to add this feature.

twyatt, 2023-07-09:
@rocketraman if you're up for creating a PR, I'd be happy to have the feature live in a separate module/artifact that consumers can optionally pull into their projects.

closed 2024-07-30 by twyatt

rocketraman, 2025-06-02:
@renaudcerrato The bulk of my implementation [is above](https://github.com/JuulLabs/kable/issues/521#issuecomment-1626588036). Its just missing the `chunked` implementation, which is here:

```kotlin
/**
 * See https://github.com/Kotlin/kotlinx.coroutines/issues/1302.
 *
 * This custom implementation forwards chunks every interval, including empty buffers.
 * To prevent memory exhaustion it will also forward chunks that exceed `maxSize`.
 */
fun <T> Flow<T>.chunked(maxSize: Int, interval: Duration, initialInterval: Duration) = channelFlow {
  val buffer = mutableListOf<T>()
  val bufferMutex = Mutex()
  var initialDelay: Duration? = initialInterval

  val flushJob = launch {
    while(isActive) {
      initialDelay?.let { delay(it) } ?: delay(interval)
      initialDelay = null
      bufferMutex.withLock {
        send(buffer.toList())
        buffer.clear()
      }
    }
  }

  collect { value ->
    bufferMutex.withLock {
      buffer.add(value)
      if (buffer.size >= maxSize) {
        send(buffer.toList())
        buffer.clear()
      }
    }
  }

  flushJob.cancelAndJoin()

  if (buffer.isNotEmpty()) {
    send(buffer.toList())
    buffer.clear()
  }
}
```

renaudcerrato, 2025-06-06:
@rocketraman : here's my `Flow<T>.window(Duration)` implementation loosely based on yours:

https://gist.github.com/renaudcerrato/230d6c8ef4e405acba43a761b4e2f555

Quick (and dirty) example:

```kotlin
    scanner.advertisements
        .window(2.seconds)
        .sample(500.milliseconds)
        .map {
            // group by Identifer
            it.groupByTo(mutableMapOf<Identifier, MutableList<Advertisement>>()) {
                it.identifier
            }
        }
        .map {
            // reduce depending on your application
            it.values.fold(emptyList<Advertisement>()) { acc, advertisements ->
                acc + advertisements.last()
            }
        }
        ...
```

rocketraman, 2025-06-10:
Thanks for sharing @renaudcerrato