JuulLabs/kable #522: State transition directly to disconnected

rocketraman, 2023-07-05:
Experimenting on Android. When I receive an advertisement, I am launching a `collect` on `peripheral.state`, and I see immediately a state of `Disconnected(status=null)`.

The peripheral has not previously been in a `Connected` state, and in fact, not even the `Disconnecting` state is seen prior to `Disconnected`, which as far as I can tell from the state diagram in the README, should not be possible.

twyatt, 2023-07-05:
Good point, the diagram doesn't properly represent that `Peripheral`'s `state` is a `StateFlow` with a default value of `Disconnected`.

The diagram should be updated and/or the default value of the `state` should be changed (perhaps to `null` — to more accurately represent the _unknown_ state).

...not quite sure if having `null` as default adds any value. Is there a use-case you're hitting where knowing that the state is unknown (rather than `Disconnected`) would be helpful?

rocketraman, 2023-07-06:
> ...not quite sure if having `null` as default adds any value. Is there a use-case you're hitting where knowing that the state is unknown (rather than `Disconnected`) would be helpful?

After I receive an advertisement for a compatible device, I have an application state called "visible", which means that I've seen the advertisement, but have not yet connected to the device. After I connect to the device, it becomes "connected". However, I think my app logic with the current setup is ok -- I attempt to differentiate "Disconnected but Visible" vs "Really Disconnected" based on the disconnected status. So maybe this is just a documentation issue.

Especially if https://github.com/JuulLabs/kable/issues/521 is resolved, then there would be no ambiguity between "Really Disconnected" and "Disconnected but Visible".

jeddchoi, 2023-08-26:
I'm also having this issue. It changes directly Connected to Disconnected(Timeout).
Because state is Disconnected before connection, I have to run some operation on Disconnecting, but can't.

twyatt, 2023-08-26:
@jeddchoi can you describe your use-case in more detail?

Are you trying to perform an action on disconnect? If so, is there a reason you can't do something to the effect of?:

```kotlin
peripheral.state
        .filterIsInstance<Disconnected>()
        .onEach {
            // todo: perform desired action on disconnect
        }.launchIn(scope)
```

jeddchoi, 2023-08-26:
I'm making seat reservation android app using beacon. There are user states such as `Reserved`, `Occupied`, `Away`, etc.
When user reserve a seat and goes close to seat which has beacon, ble connection is established and change user state to `Occupied`. And if user leaves away from the seat and ble connection is disconnected, user state is changed to `Away`.

So this use case is that I need to some operation when ble connection is disconnected after connection.
Because `peripheral.state` default state is `Disconnected`, I can't do operation on `Disconnected`.

> peripheral.state
        .filterIsInstance<Disconnected>()
        .onEach {
            // todo: perform desired action on disconnect
        }.launchIn(scope)

I think your above suggesstion would not distinguish between first `Disconnected` before any connection and `Disconnected` after connection. 
I would be grateful if you change default state to some kind of state like `Unknown` as you said.

jeddchoi, 2023-08-27:
After more investigation, I found that its state goes like `Disconnected(null)` -> `Connecting.Bluetooth` -> `Disconnected(133)` -> `Connecting.Bluetooth` -> `Disconnected(133)` ->  ... -> `Connecting.Bluetooth` -> `Connecting.Services` -> `Connecting.Observes` -> `Connected`.

I think I need some flag or `Disconnected.Status` to show whether it was connected successfully or not.