JuulLabs/kable #540: NullPointerException: characteristic.value must not be null

twyatt, 2023-07-27:
> I'm getting several reports of this crash:
> 
> ```
> Fatal Exception: java.lang.NullPointerException: characteristic.value must not be null
>        at com.juul.kable.gatt.Callback.onCharacteristicRead(Callback.kt:144)
>        at android.bluetooth.BluetoothGatt$1$6.run(BluetoothGatt.java:405)
>        at android.os.Handler.handleCallback(Handler.java:938)
>        at android.os.Handler.dispatchMessage(Handler.java:99)
>        at android.os.Looper.loopOnce(Looper.java:201)
>        at android.os.Looper.loop(Looper.java:288)
>        at android.os.HandlerThread.run(HandlerThread.java:67)
> ```

_Originally posted by @miwright2 in https://github.com/JuulLabs/kable/issues/485#issuecomment-1650359561_

twyatt, 2023-07-27:
@miwright2 is this isolated to specific versions of Android?

miwright2, 2023-07-27:
@twyatt -- this is the report from the last 30 days.  It looks spread between Android 9, Android 11, and Android 12.  The device types are below:
[image]

twyatt, 2023-07-27:
@miwright2 I have a suspicion of what is going wrong.

My guess is that during a BLE I/O operation there was an error result (and to my surprise) Android may be providing `null` for the data `ByteArray` in the event an error occurred.

Unfortunately the error status is logged _after_ trying to create the object that is throwing the NPE:

https://github.com/JuulLabs/kable/blob/0a09c9889c7ec3d45d77b7925ccacf8a95555777/core/src/androidMain/kotlin/gatt/Callback.kt#L154-L160

I'll think about a good way to address this and try to get you a SNAPSHOT soon to try.

twyatt, 2023-07-27:
@miwright2 what version of Kable are you currently on? Asking to know what version I should work off of for providing you a SNAPSHOT (so that it has to smallest number of possible changes from your actively used version).

miwright2, 2023-07-27:
@twyatt I am on the previous snapshot with the ANR fix.  I believe that was based on 0.23.  I could try moving to 0.25 if you think that's a good idea.

twyatt, 2023-07-27:
If you're able to bump up to 0.25.0 then I'll make the snapshot based on that version. It would be preferred simply because other fixes may have made it into the latest version.

miwright2, 2023-07-27:
Let's do that approach and base it off of 0.25.  Thanks for the support!

twyatt, 2023-07-28:
Pretty sure I found the cause of the crash: when upgrading the Android compile SDK (#455) the Android BLE callbacks got new I/O callbacks (these new callbacks do not allow `null` for the data) whereas the old deprecated callbacks sometimes provide `null` for I/O data — the old callbacks were forwarding to the new callbacks (for backwards compatibility).

twyatt, 2023-07-28:
```
repositories {
    maven("https://oss.sonatype.org/content/repositories/snapshots")
}

dependencies {
    implementation("com.juul.kable:core:0.25.0-issue-540-1-SNAPSHOT")
}
```

closed 2023-08-01 by twyatt

twyatt, 2023-08-01:
This should be fixed in 0.25.1.

PR #541 (2023-08-01): Properly handle `null`s in deprecated Android BLE callbacks
Closes #540