JuulLabs/kable #544: Crash logs from Android on cancellation

rocketraman, 2023-08-01:
I am getting crash logs from Android due to cancellation.

This is the stack I see:

```
 va.e0
	at com.juul.kable.BluetoothDeviceAndroidPeripheral$read$$inlined$execute$1.invokeSuspend(Connection.kt:21)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:8)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:107)
	at android.os.Handler.handleCallback(Handler.java:883)
	at android.os.Handler.dispatchMessage(Handler.java:100)
	at android.os.Looper.loop(Looper.java:214)
	at android.os.HandlerThread.run(HandlerThread.java:67)
	Suppressed: kotlinx.coroutines.m0: [v1{Cancelling}@5439012, Dispatchers.Main]
```

rocketraman, 2023-08-01:
> What version of Kable is this on?

Version 0.24.0.

twyatt, 2023-08-03:
I'm not sure if the stacktrace is misleading, but the first line of it points to `Connection.kt:21`:

https://github.com/JuulLabs/kable/blob/f71e085fb0182ca43a07c88693d641dceb0f8974/core/src/androidMain/kotlin/Connection.kt#L21-L23

I normally wouldn't expect this exception to be thrown on cancellation:

https://github.com/JuulLabs/kable/blob/f71e085fb0182ca43a07c88693d641dceb0f8974/core/src/androidMain/kotlin/Connection.kt#L94-L99

Instead I might expect it to be thrown on a subsequent I/O operation after a previous I/O cancellation (if there is a bug in Kable handling of the cancellation).

Are there specific steps you're doing to reproduce it (or, more specifically, how is the cancellation occurring? Is there surrounding code around the code snippet you shared that performs the cancellation?

rocketraman, 2023-08-05:
> Are there specific steps you're doing to reproduce it (or, more specifically, how is the cancellation occurring)? Is there surrounding code around the code snippet you shared that performs the cancellation?

I'm actually not sure how to reproduce it. Its an error I see periodically pop up in Bugfender which we use to track errors on production devices.

I don't do any cancellation of the read/write scopes myself. I do cancellation of the peripheral scanning scope in certain cases, and I call the `peripheral.disconnect()` method if requested by the user.

rocketraman, 2023-08-15:
Well, I'm not sure either. I think I have some wonky code related to peripheral state tracking which may be the cause of the issue. I'll try to refactor a bit to see if the problem goes away, or if the problem continues to happen, I can get a clearer idea of where the cancellation is coming from.

twyatt, 2023-10-04:
@rocketraman when you have a chance, can you test [`0.28.0-rc`](https://github.com/JuulLabs/kable/releases/tag/0.28.0-rc) and let me know if the stacktrace has changed for this issue?

rocketraman, 2023-10-04:
I actually haven't seen this happen in a while, so I suspect the fix of the wonky code I mentioned before resolved it. I'll go ahead and close. Thanks!

closed 2023-10-04 by rocketraman

twyatt, 2023-10-04:
Great to hear. Thanks for following up. 👍