JuulLabs/kable #547: For a KMP project, XCode cannot build the project after version 0.25.0

privateAUser, 2023-08-09:
It works fine until version 0.24.0 . but after version 0.25.0, although I can get build in android studio, I can't get build in xcode. it gives an error as below.

Could not find "/Users/.../.gradle/caches/modules-2/files-2.1/com.juul.kable/core-iosx64/0.25.1/13734d461661cdfa38d3ec3707006fe5fc970779/core.klib" in [/Users/.../Desktop/.../iosApp/Pods, /Users/.../.konan/klib, /Users/user/.konan/kotlin-native-prebuilt-macos-x86_64-1.8.22/klib/common, /Users/.../.konan/kotlin-native-prebuilt-macos-x86_64-1.8.22/klib/platform/ios_x64]

twyatt, 2023-08-09:
Could you provide a small reproducer, please?

mehmetakify, 2023-08-12:
I encountered same issue and this is the small reproducer:

https://github.com/mehmetakify/Reproducer-Kable

You can change Kable version in this file:
https://github.com/mehmetakify/Reproducer-Kable/blob/main/buildSrc/src/main/java/Dependencies.kt

Kable version is 0.25.1 and XCode build is failed with the same error. After changing to 0.24.0 XCode build is successed.

twyatt, 2023-08-13:
Thanks @mehmetakify, I'll take a look soon.

twyatt, 2023-08-14:
When I tried your project I received the following error:

```shell
./gradlew podPublishDebugXCFramework

> Task :shared:compileKotlinIosArm64 FAILED
w: skipping /Users/twyatt/.gradle/caches/modules-2/files-2.1/com.juul.kable/core-iosarm64/0.25.1/e138fdc1cb3fdee08a1008ea313b48eeb42f9f63/core.klib. Incompatible abi version. The current default is '1.7.0', found '1.8.0'. The library produced by 1.9.0 compiler
e: Could not find "/Users/twyatt/.gradle/caches/modules-2/files-2.1/com.juul.kable/core-iosarm64/0.25.1/e138fdc1cb3fdee08a1008ea313b48eeb42f9f63/core.klib" in [/Users/twyatt/Projects/Reproducer-Kable, /Users/twyatt/.konan/klib, /Users/twyatt/.konan/kotlin-native-prebuilt-macos-aarch64-1.8.22/klib/common, /Users/twyatt/.konan/kotlin-native-prebuilt-macos-aarch64-1.8.22/klib/platform/ios_arm64]
```

I changed Kotlin to `1.9.0` in your `Dependencies.kt` and it worked and compiled in Xcode. Let me know if that resolves the issue for you.

mehmetakify, 2023-08-14:
The new library linkage of Kotlin 1.9.0 solved the problem. Thank you!

twyatt, 2023-08-14:
Thanks for confirming @mehmetakify.

closed 2023-08-14 by twyatt