JuulLabs/kable #554: NSInternalInconsistencyException on foreground scan on iOS

mtrewartha, 2023-08-22:
After updating to v0.26.0, an `NSInternalInconsistencyException` seems to be thrown at `CentralManager.kt:43:36` with the following message:

>State restoration of CBCentralManager is only allowed for applications that have specified the \"bluetooth-central\" background mode"

We don't scan in the background, so we don't specify a background mode. I'm guessing this is related to #552? For now, we'll just hold off on the update as we don't need it.

twyatt, 2023-08-22:
Thanks for reporting. It seems we need to make that functionality opt-in. 👍

twyatt, 2023-08-22:
This will likely require some significant Kable API changes. I'm hopeful I'll be able to work on it soon and slate it for the next Kable release.

mtrewartha, 2023-08-23:
Good to know and thanks for the quick response!

twyatt, 2023-09-06:
> This will likely require some significant Kable API changes.

Ended up going with a solution (#561) that did not change the API significantly.

PR #561 (2023-09-06): Allow `CentralManager` to be configured
Apple consumers can now _optionally_ configure the internally used global `CentralManager`:

```kotlin
CentralManager.configure {
    stateRestoration = true // `false` by default (when not specified).
}
```

This makes "background support" (state restoration) **opt-in** — in other words: not calling `configure` will not enable state restoration.

_As stated in the [updated documentation](https://github.com/JuulLabs/kable/tree/twyatt/central-manager-initialize#background-support), `configure` must be called before using most of Kable's functionality._

C [... cut]
closed 2023-09-06 by twyatt