JuulLabs/kable #567: Worker never ends if I use kable in it

solvek, 2023-09-13:
If I just create a `periferal` instance inside worker then this worker never ends. 

I am enqueueing a unique work. For the first time it runs fine (I see in logs `Finished worker`), when I enqueue again the work it does not start. I assume this is because the previous work was not finished.

```kotlin

@HiltWorker
class DfuWorker @AssistedInject constructor(@Assisted private val context: Context,
                                            @Assisted workerParameters: WorkerParameters
) : CoroutineWorker(context, workerParameters) {
    override suspend fun doWork(): Result = withContext(Dispatchers.IO) {
        Timber.tag(TAG).d("Started")

        // If I comment this line all works as expected
        val peripheral = peripheral(Biosensors.bluetoothDevice(deviceId))

        Timber.tag(TAG).d("Finished worker")
        Result.success()
    }

    companion object {
        private const val TAG = "DfuWorker"

        fun WorkManager.runDfuWorker(deviceId: String, deviceType: DeviceType){
            val data = Data
                .Builder()
                .putString(DATA_DEVICE_ID, deviceId)
                .putInt(DATA_DEVICE_TYPE, deviceType.code)
                .build()

            val workRequest = OneTimeWorkRequest
                .Builder(DfuWorker::class.java)
                .addTag("FIRMWARE_UPDATE")
                .setInputData(data)
                .build()

            enqueueUniqueWork(
                "update_firmware",
                ExistingWorkPolicy.KEEP,
                workRequest
            )

            Timber.tag(TAG).d("Requested enqueue worker for dfu.")
        }
    }
}
```

solvek, 2023-09-13:
I have the most recent version 0.27.0

twyatt, 2023-09-13:
This is likely a duplicate of #378.

I've been slowly integrating changes in Kable to address the issue, it will be some time before it can be completely fixed.

In the mean time, one possible workaround is to create a parent `Job` for your peripheral that you explicitly control its lifecycle, for example:

```kotlin
override suspend fun doWork(): Result = withContext(Dispatchers.IO) {
    val scope = CoroutineScope(Job(coroutineContext[Job]))
    try {
        val peripheral = scope.peripheral(..)
    } finally {
        scope.cancel()
    }
}
```

twyatt, 2023-09-25:
I'm hoping to get changes into [0.29.0](https://github.com/JuulLabs/kable/milestone/44) that would remove the need for this workaround.

twyatt, 2024-04-18:
> I'm hoping to get changes into [0.29.0](https://github.com/JuulLabs/kable/milestone/44) that would remove the need for this workaround.

Unfortunately lots of other things came up and this didn't happen.
It'll likely be a bit more effort than originally anticipated, so it will be a number of version from now before this is fully resolved. #612 is a small step in the direction needed to fully resolve this, but I haven't had time to test it enough.

I'll try to get back to this when I can, but being that there is a workaround, it might not get prioritized very high vs. some other bugs. Thanks for being patient!

closed 2024-09-25 by twyatt