JuulLabs/kable #568: Is it possible to turn notifications off?

ercan, 2023-09-13:
Hi, is it possible to turn the notifications off without disconnecting the peripheral? I tried to do some reflection but since Connection class is internal I couldn't get define a variable for that.

twyatt, 2023-09-13:
Are you saying you want an `observe` active but **not** having the underlying characteristic have its notification enabled?

Can you describe a little more about your use case, please?

ercan, 2023-09-13:
I mean `setCharacteristicNotification( ..., true)` is called when we do `observe` under the hood, is it possible to call `setCharacteristicNotification(..., false)` somehow by using kable over the `observe` flow or somehow?

I actually want to force stop `observe`, and if I need that flow later, I'd start `observe` again.
My App has a button to simply start and stop data observation and my peripheral has an LED to indicate if the notification is set true or false. Once I click the stop button, I just need my peripheral to know `setCharacteristicNotification` is set as false 🙂

ercan, 2023-09-13:
Uh. I simply used `takeWhile { start/stop flag }` after the observe call and it did the trick 👍

closed 2023-09-13 by ercan