JuulLabs/kable #610: Feature request: Parcelable Advertisement on Android

eddieSullivan, 2023-12-12:
I wonder if you would consider making the `AndroidAdvertisement` implement `Parcelable`?

The [Android native ScanResult](https://developer.android.com/reference/android/bluetooth/le/ScanResult) implements the Parcelable interface, so I think this should be possible, though I don't know if there are internal details of the Advertisement implementation that make it more difficult.

The use case is an android view that shows a list of devices in range, implemented using Jetpack Compose. I would like to be able to seamlessly handle app configuration changes using `rememberSaveable`.

Yes, I could store the list of advertisements in a global singleton, outside of the view hierarchy, but that feels less in line with Android architectural idioms.

eddieSullivan, 2023-12-12:
And the same thing would be nice for Peripheral. [Android's BluetoothDevice](https://developer.android.com/reference/android/bluetooth/BluetoothDevice) implements Parcelable.

twyatt, 2023-12-12:
Sounds like a good feature to have. 👍

I'm currently pretty busy on other projects at the moment, but I'll try to find some time to work on this.

PRs are also welcome. 😁

twyatt, 2024-03-10:
> And the same thing would be nice for Peripheral. [Android's BluetoothDevice](https://developer.android.com/reference/android/bluetooth/BluetoothDevice) implements Parcelable.

Having Kable's `AndroidAdvertisement` be `Parcelable` can be done, but unfortunately: I don't see a feasible way to do the same for `AndroidPeripheral`. As `AnroidPeripheral` is stateful (holds onto a connection) which cannot be written to a `Parcel`. This was done to make a peripheral have a more intuitive API.

Android's `BluetoothDevice` is `Parcelable` because it is stateless (is a thin wrapper around a BLE address), and from that stateless object a `BluetoothGatt` is created (which holds the underlying connection).

twyatt, 2024-03-12:
@eddieSullivan, #652 makes `AndroidAdvertisement` `Parcelable`.

You can test it with:

```kotlin
repositories {
    maven("https://oss.sonatype.org/content/repositories/snapshots")
}

dependencies {
    implementation("com.juul.kable:core:0.29.1-issue-610-1-SNAPSHOT")
}
```

closed 2024-03-21 by twyatt

twyatt, 2024-03-21:
Shipped in [0.30.0](https://github.com/JuulLabs/kable/releases/tag/0.30.0).

PR #652 (2024-03-21): Make `AndroidAdvertisement` parcelable
Closes #610.