JuulLabs/kable #613: Android: Deadlock when Read RSSI Doesn't Callback

miwright2, 2023-12-21:
- Pixel 7, current version.  
- Running local build of Kable from HEAD of `main`.

I'm encountering what appears to be a deadlock when I have two active connections (different peripheral devices) that poll on the remote RSSI.  It looks to me like Android can fail to callback with the remote RSSI.  When that occurs, Kable waits indefinitely for that callback, blocking subsequent synchronous operations over the connection (e.g. `peripheral.write`). Note that Kable is still providing our app with updates from notifications (e.g. `onCharacteristicChanged` from Android).

Please let me know if there is additional information that I can provide.

twyatt, 2023-12-21:
Does this only happen when you have 2 (or more) active peripheral connections? In other words, do you ever see this issue with only a single peripheral connected?

miwright2, 2023-12-21:
I don't think so.

miwright2, 2024-01-02:
@twyatt -- happy new year!  we're starting to plan around a feature that will need to read RSSI while connected.  I just wanted to touch base to see what you thought about the timing for this.  We can fork and dig into it, but would only really want to do that if you think this bug is further out.

twyatt, 2024-01-02:
Happy New Year to you as well! 😁

I have two big-ish projects for work that take priority, so I'm guessing I won't have a lot of time to dedicate to Kable for at least a month.

I try to find time off hours to patch bugs, but this one may require a bit of investigation. Anything you can do to add clarity to the issue would be super helpful.

I'll try to find time sooner, but no promises.

twyatt, 2025-02-08:
Sorry I haven't given this the attention is deserves.
I'm curious though, after many recent refactors in Kable, is this still an issue is the latest versions?

Jaco233, 2025-11-09:
Hey @twyatt 
I’ve done some testing and was able to reproduce the issue on iOS.

ramosgerald255, 2026-11-08:
Hey @twyatt We're still encountering this issue on Android. We are currently using 0.45.0 version and it seems the Rssi status is blocking some write operation on our devices. Is there any update on this issue?