JuulLabs/kable #632: Update `GattRequestRejectedException` with error code from `writeCharacteristic`

ArtemBurmistrov, 2024-02-15:
In Android 13 the [`BluetoothGatt.writeCharacteristic`](https://developer.android.com/reference/kotlin/android/bluetooth/BluetoothGatt#writecharacteristic) function with `boolean` return type was deprecated in favor of [a similar one](https://developer.android.com/reference/kotlin/android/bluetooth/BluetoothGatt#writecharacteristic_1) with `int` return type.
It would be beneficial if `GattRequestRejectedException` could be updated with additional error codes so library users can have more context on why an exception was thrown.

twyatt, 2024-02-15:
Thanks for pointing this out, and linking to the relevant APIs. 👍 
Seems like a quick/easy addition, I'll try to get to this soon.

nooraujla90, 2024-02-20:
https://juullabs.atlassian.net/browse/CA-2870

twyatt, 2024-03-12:
@ArtemBurmistrov, #648 provides the error type via an exception property. Can you try it, and let me know if it fits your needs?:

```kotlin
repositories {
    maven("https://oss.sonatype.org/content/repositories/snapshots")
}

dependencies {
    implementation("com.juul.kable:core:0.29.1-issue-632-1-SNAPSHOT")
}
```

ArtemBurmistrov, 2024-03-18:
Hey @twyatt, that looks awesome, thank you! I checked PR and it looks like this is exactly what I need.

closed 2024-03-21 by twyatt

twyatt, 2024-03-21:
Shipped in [0.30.0](https://github.com/JuulLabs/kable/releases/tag/0.30.0).

PR #648 (2024-03-21): Provide write failure as property of exception
Introduces `GattWriteException` which may be thrown from `AndroidPeripheral.write`[^1] functions (Android-only). The `GattWriteException` has a `result` property which contains the failure enum.

Closes #632

[^1]: When writing characteristic or descriptor.