JuulLabs/kable #634: Help with iOS implementation

Codegn, 2024-02-19:
Hello everyone,

First of all, thank you for the great work with Kable.

We have been successfully using Kable in an Android app, but we are at a bit of a loss trying to implement the same features in iOS. I was hoping that you could guide us through the steps that are necessary to use Kable with iOS since the examples provided are made for Android only.

Best regards!

twyatt, 2024-02-21:
Unfortunately at this time I don't have the bandwidth to create an iOS example.

Perhaps you could try asking for pointers either in the [discussions](https://github.com/JuulLabs/kable/discussions) or on the `#juul-libraries` [kotlinlang](https://kotlinlang.org/community/) Slack?

twyatt, 2024-08-08:
Closing as duplicate of https://github.com/JuulLabs/sensortag/issues/11.

closed 2024-08-08 by twyatt