JuulLabs/kable #637: Bluetooth connection lost not recognized only in iOS in KMP project

cksgud, 2024-02-29:
Hi 

I have an issue with the Kable library version 0.27.0 in a KMP project.

On connection loss an exception is only thrown in Android, but never in iOS.

See the following example code:

try {

     peripheral.connect()

 } catch (cl: ConnectionLostException) {

     // Android reaches this on connection loss - but iOS does not 

 } 

Still I could see the following in the console log for my connected iOS 16.5.1 iPhone 8 Plus:

Debug: (Kable) F5F6DD67-6A8D-2ED9-4649-BCBD21B9069E CentralManagerDelegate state change

  state: DidDisconnect(identifier=F5F6DD67-6A8D-2ED9-4649-BCBD21B9069E, error=Error Domain=CBErrorDomain Code=6 "The connection has timed out unexpectedly."

Do you have an idea how I can detect connection loss for both platforms reliable?

Thank you so much for this great library

twyatt, 2024-02-29:
I think the proper error handling around connecting in iOS is missing (`todo`s in the code). I have time allocated in the coming weeks to work on Kable, I'll try to implement/fix this at that time.

Thanks for reporting the issue!

muellnes, 2024-04-11:
Hi @twyatt 

we already are affected by this issue - do you have any plans on implementing it with the next version?

Thank you

twyatt, 2024-04-11:
I'll try to find some time soon. 👍

twyatt, 2024-04-15:
I don't currently have the means of testing this, but if either @cksgud or @muellnes could try this `SNAPSHOT`, and report back if it works (and the logs/exception produced), I'd be very much appreciative:

```kotlin
repositories {
    maven("https://oss.sonatype.org/content/repositories/snapshots")
}

dependencies {
    implementation("com.juul.kable:core:0.31.0-issue-637-1-SNAPSHOT")
}
```

muellnes, 2024-04-17:
After a long test period, with multiple test cases, the issue does not appear anymore and it fixes the problem that BLE disconnection is not recognised. 

Thank you so much for the changes, hope we can make it in the next release version 🥇

twyatt, 2024-04-17:
Thanks so much for confirming @muellnes!

Any chance you can provide some logging around the failure case, and the exception that results? _I want to make sure that things propagated as I intended._

I can probably get a release cut in the coming days with this change. 👍

muellnes, 2024-04-17:
Hey @twyatt. This should be no problem. I will try to get you the run logs for the error case as quick as possible :)

twyatt, 2024-04-18:
Thanks for the logs! They helped confirm the messaging that I was hoping to convey in the exception message. 💯

twyatt, 2024-04-18:
Released in [0.31.1](https://github.com/JuulLabs/kable/releases/tag/0.31.1).

PR #668 (2024-04-18): Watch for connection loss during connect
Closes #637