JuulLabs/kable #653: peripheral.observe  didUpdateNotificationStateForCharacteristic

DayDayUpAllInOne, 2024-03-10:
When I use the code below,The kable library reports an error
`actual suspend fun notify(device: BleDeviceEntity): Flow<ByteArray> {
    val notifyCharacteristic = characteristicOf(
      uuidFrom(bleUuid.serviceUuid).toString(),
      uuidFrom(bleUuid.notifyUuid).toString()
    )
    return device.peripheral.observe(notifyCharacteristic)
  }`
error message:
W/Kable/Delegate: 890FF1DD-33D3-AF4F-2CD8-7340E8C3B112 890FF1DD-33D3-AF4F-2CD8-7340E8C3B112 didUpdateNotificationStateForCharacteristic
  service: 8C858000-0302-41C5-B46E-CF057C562025
  characteristic: 8C858000-0302-4D46-BEC2-EE689169F628
  error: Error Domain=CBATTErrorDomain Code=10 "The attribute could not be found." UserInfo={NSLocalizedDescription=The attribute could not be found.}
[image]

How to solve this problem? Thanks!

twyatt, 2024-03-11:
What version of Kable are you using?

DayDayUpAllInOne, 2024-03-11:
> What version of Kable are you using?

hi twyatt, the Kable version I use is 0.24.0, this problem occurs on iOS, Android is normal

DayDayUpAllInOne, 2024-03-11:
I added the observationExceptionHandler method, and the program can receive the notification data from the Bluetooth device normally.
`val peripheral = defaultScope.peripheral(it) {
            onServicesDiscovered {
              bleLogger.i("""BleClient/startScan/onServicesDiscovered:${it.peripheralName}""")
            }
            observationExceptionHandler { cause ->
              // Log failure instead of propagating associated `observe` flow.
              bleLogger.i("""BleClient/startScan/observationExceptionHandler:$cause""")
            }
          }
`
Log:
`
 Runner[1678:356710] W/Kable/Delegate: 890FF1DD-33D3-AF4F-2CD8-7340E8C3B112 890FF1DD-33D3-AF4F-2CD8-7340E8C3B112 didUpdateNotificationStateForCharacteristic
  service: 8C858000-0302-41C5-B46E-CF057C562025
  characteristic: 8C858000-0302-4D46-BEC2-EE689169F628
  error: Error Domain=CBATTErrorDomain Code=10 "The attribute could not be found." UserInfo={NSLocalizedDescription=The attribute could not be found.}
>>> info#ESIot/ble <<<BleClient/startScan/observationExceptionHandler:com.juul.kable.IOException: Error Domain=CBATTErrorDomain Code=10 "The attribute could not be found." UserInfo={NSLocalizedDescription=The attribute could not be found.}/1710152968286
`

Why does the program become normal after adding the observationExceptionHandler method?

twyatt, 2025-01-19:
@DayDayUpAllInOne based on your nRF Connect screenshot, I believe your peripheral is missing the "Client Characteristic Configuration Descriptor" (`00002902-0000-1000-8000-00805F9B34FB`) under your `8C858000-0302-4D46-BEC2-EE689169F628` characteristic (similar to https://github.com/dariuszseweryn/RxAndroidBle/issues/202#issuecomment-304650959).

Generally, when a characteristic is observed, the "Client Characteristic Configuration Descriptor" needs to be written to in order to enable to indication or notification of the associated characteristic.

On Android, Kable handles this for you, but if the "Client Characteristic Configuration Descriptor" is missing, Kable simply logs a warning.

On iOS, the system handles writing to "Client Characteristic Configuration Descriptor" automatically, and as you've found: if it cannot, it results in a failure which Kable propagates.

As you've seen, the behavior between the two platforms is not consistent. I'm inclined to change Android (to be in parity w/ iOS) to also raise an exception if the "Client Characteristic Configuration Descriptor" does not exist.

Ultimately, you should have the firmware fixed on your peripheral to provide the "Client Characteristic Configuration Descriptor" for any characteristics that you intend to observe.