JuulLabs/kable #654: Eventual ANR when consuming perpetual advertisements

djweber, 2024-03-11:
Still trying to figure out this one. I've been running into an ANR that I seem to consistently hit after anywhere from 15-45 minutes of successful scanning. I've been testing on an Galaxy S22 running Android 13. The ANR is as follows:

```Cmd line: <package>
Build fingerprint: 'samsung/b0qsqw/b0q:13/TP1A.220624.014/S908USQS2BWA2:user/release-keys'
ABI: 'arm64'
Build type: optimized
suspend all histogram:	Sum: 41.270ms 99% C.I. 0.494us-948.800us Avg: 78.311us Max: 2379us
DALVIK THREADS (168):
"main" prio=5 tid=1 TimedWaiting
  | group="main" sCount=1 ucsCount=0 flags=1 obj=0x7334ef58 self=0xb4000079a6fc6380
  | sysTid=6202 nice=0 cgrp=background sched=0/0 handle=0x7b521c3500
  | state=S schedstat=( 427635411952 12382667868 187478 ) utm=34641 stm=8121 core=2 HZ=100
  | stack=0x7fceb7f000-0x7fceb81000 stackSize=8188KB
  | held mutexes=
  at jdk.internal.misc.Unsafe.park(Native method)
  - waiting on an unknown object
  at java.util.concurrent.locks.LockSupport.parkNanos(LockSupport.java:252)
  at kotlinx.coroutines.BlockingCoroutine.joinBlocking(Builders.kt:88)
  at kotlinx.coroutines.BuildersKt__BuildersKt.runBlocking(Builders.kt:59)
  at kotlinx.coroutines.BuildersKt.runBlocking(unavailable:1)
  at kotlinx.coroutines.BuildersKt__BuildersKt.runBlocking$default(Builders.kt:38)
  at kotlinx.coroutines.BuildersKt.runBlocking$default(unavailable:1)
  at kotlinx.coroutines.channels.ChannelsKt__ChannelsKt.trySendBlocking(Channels.kt:38)
  at kotlinx.coroutines.channels.ChannelsKt.trySendBlocking(unavailable:1)
  at com.juul.kable.BluetoothLeScannerAndroidScanner$advertisements$1$callback$1.onBatchScanResults(BluetoothLeScannerAndroidScanner.kt:54)
  at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$2.run(BluetoothLeScanner.java:681)
  at android.os.Handler.handleCallback(Handler.java:942)
  at android.os.Handler.dispatchMessage(Handler.java:99)
  at android.os.Looper.loopOnce(Looper.java:226)
  at android.os.Looper.loop(Looper.java:313)
  at android.app.ActivityThread.main(ActivityThread.java:8741)
  at java.lang.reflect.Method.invoke(Native method)
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:571)
  at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1067)
```

This can be followed to this line in `BluetoothLeScannerAndroidScanner.kt:54`:

https://github.com/JuulLabs/kable/blob/b69d0ff84e0a5e1075c695d35bb7de32f364280d/core/src/androidMain/kotlin/BluetoothLeScannerAndroidScanner.kt#L54

For context, I have an ongoing BLE scan running in a foreground service to connect to certain devices as they appear. This scan is meant to run alongside the app, indefinitely.

twyatt, 2024-03-11:
What version of Kable and Kotlin coroutines are you using?
I haven't investigate yet, but wondering if this could be related to https://github.com/Kotlin/kotlinx.coroutines/issues/3940?

djweber, 2024-03-11:
I am on Kable 0.29.0 and coroutines version 1.7.3:

```
implementation("org.jetbrains.kotlinx:kotlinx-coroutines-core:1.7.3")
implementation("org.jetbrains.kotlinx:kotlinx-coroutines-android:1.7.3")
```

I mean to mention that I have seen this ANR happen on a Pixel 7a as well. I noticed this issue while digging around earlier:

https://github.com/Kotlin/kotlinx.coroutines/issues/3952

djweber, 2024-03-11:
Any chance this could be due to buffering? Was looking at the [callbackFlow](https://kotlinlang.org/api/kotlinx.coroutines/kotlinx-coroutines-core/kotlinx.coroutines.flow/callback-flow.html) docs and it mentioned ways to prevent blocking:

```// To avoid blocking you can configure channel capacity using
            // either buffer(Channel.CONFLATED) or buffer(Channel.UNLIMITED) to avoid overfill```

I'm running a scan with a balanced power mode setting and no delivery delay.

djweber, 2024-03-13:
@twyatt fairly certain this was due to a deadlock I was creating upstream, so I am going to close this for now. I will let you know if I encounter it again.

twyatt, 2024-03-13:
Sounds good. Thanks @djweber for following up. 👍

djweber, 2024-05-21:
@twyatt I am not entirely convinced that this issue is completely eliminated. Similar to https://github.com/JuulLabs/kable/issues/485, I have seen multiple ANRs on our test track tracing back to this line: https://github.com/JuulLabs/kable/blob/main/core/src/androidMain/kotlin/BluetoothLeScannerAndroidScanner.kt#L50. I am currently on Kable 0.31.1. 

To recap, I have a foreground service that can run a scan either periodically or on an interval (this ANR has been observed on both setups). I had originally thought I had fixed this issue due to a blocked subscriber downstream. My Kable advertisement flow maps the advertisements to a loosely wrapped `Peripheral` abstraction.

I have only ever seen this happen once on my end, but there are definititely users encountering it. I was considering forking the repo to have an additional scanner implementation that doesn't use the `callbackFlow`. Not sure if this is viable though.

```
scanner
    .advertisements
    .conflate()
    .catch {
        throw BleException(it.message, it.cause)
    }
    .map { advertisement ->
        cachedDevices.computeIfAbsent(advertisement.address) {
            // maps Kable advertisement + peripheral to a `Device` abstraction
            KableDeviceFactory(scanOptions.shouldAutoConnect)
                .create(coroutineScope, advertisement).also {
                    it.rssi = advertisement.rssi
                }
        }.apply {
            lastAdvertisedAt = LocalDateTime.now()
        }
    }
    

    
    // KableDeviceFactory
    ...
    val peripheral = coroutineScope.peripheral(advertisement) {
        transport = Transport.Le

        if (BuildConfig.DEBUG) {
            logging {
                engine = SystemLogEngine
                level = Logging.Level.Events
                format = Logging.Format.Multiline
                data = Hex
            }
        }

        onServicesDiscovered {
            /**
                * Android 14 automatically requests [Connector.TARGET_MTU], but we can request the
                * highest here for consistency with older OS versions
                */
            requestMtu(Connector.TARGET_MTU)

            device.readDeviceInfo()
        }

        autoConnectIf {
            shouldAutoConnect
        }
    } as AndroidPeripheral

    ...
```
All my downstream collector does is collect this peripheral abstraction and emit it to a flow that has plenty of extra buffer capacity and a `DROP_OLDEST` strategy:

djweber, 2024-05-21:
@twyatt It looks like this issue is still happening for me. I am using a light abstraction over Kable 0.31.1 with a perpetual scan in a foreground service, and I still see ANRs coming through Crashlytics for both batched and unbatched advertisements at both `onScanResult ` and `onBatchScanResults`. I saw https://github.com/JuulLabs/kable/pull/537 had a similar situation, which was fixed by removing the `flowOn` call.

djweber, 2024-05-22:
Also wondering if running the scan in intervals might be worth trying here.

twyatt, 2024-05-22:
I'll share my understanding of how Android BLE scanning works and hopefully it helps you track down your issue:

The [callback that is registered w/ Android for BLE scanning](https://github.com/JuulLabs/kable/blob/616488b309aab68be24006ad07461854767779c5/core/src/androidMain/kotlin/BluetoothLeScannerAndroidScanner.kt#L83) is generally called from an Android [binder thread](https://stackoverflow.com/a/15225390) — for some phones, the main thread may be used as the binder thread. 🙈 

This happens regardless of Kable being used, or if implementing a vanilla Android BLE scanner, so you have to be careful not to block too long within the scanner callback functions (e.g. [`onScanResult`](https://developer.android.com/reference/android/bluetooth/le/ScanCallback#onScanResult(int,%20android.bluetooth.le.ScanResult))). It is an unfortunate footgun.

The [`trySendBlocking`](https://github.com/JuulLabs/kable/blob/616488b309aab68be24006ad07461854767779c5/core/src/androidMain/kotlin/BluetoothLeScannerAndroidScanner.kt#L40) in Kable will block if your flow consumer (more specifically: the consumer of the `Scanner.advertisements` flow) is slow.

For example:

```kotlin
val scanner = Scanner { .. }
scanner.advertisements.collect { delay(10.seconds) } // This will block the Android binder thread in the scanner callback for ~10 seconds.
```

The `flowOn` operator, unfortunately won't be much help here (it could actually make things _slightly_ worse, as now you're also slowing the consumer slightly by having to perform context/dispatcher switches):

```kotlin
val scanner = Scanner { .. }
scanner.advertisements.map {
    delay(10.seconds)
    ExampleClass(it)
}.flowOn(Dispatchers.IO) // This will only cause the operators ABOVE this one to use Dispatchers.IO (i.e. the `map` operation) — it will have no affect (re: threading) on the `trySendBlocking` in the callback (internally in Kable)
.collect { .. }
```

This problem isn't unique to Kable, if you have an expensive operation occurring within the scanner callback, you'll also block it for too long. Without Coroutines and/or Kable, you'd traditionally have to push work to another thread as quickly as possible — it can get pretty complex, thankfully Coroutines simplifies this a lot.

So, what can _you_ do then? The best thing you can do is make sure that the consumer side is **fast**.

One option is to buffer:

```kotlin
val scanner = Scanner { .. }
scanner.advertisements
    .buffer(capacity = <big_number_here>)
    .map {
        // You can have an expensive/long running operation here (as long as your buffer doesn't fill up).
    }
    .collect { .. }
```

If the `map` is slow, then the buffer will allow the flow to continue processing upstream advertisements (with the hope that on the consumer side things catch up). If the consuming side (`map` in this example) is continually slow (then eventually the buffer will fill up and put "back pressure" on the `trySendBlocking` and block, resulting in the ANR if blocked for too long).

If you don't care about reading **every** advertisement, then `conflate` would be your best option, as it will drop older advertisements if the consuming side is too slow:

```
val scanner = Scanner { .. }
scanner.advertisements
    .conflate()
    .map {
        // You can have an expensive/long running operation here (advertisements will just be dropped until we're done).
    }
    .collect { .. }
```

Hope that helps. 🤞

djweber, 2024-05-22:
@twyatt Thank you for that overview, great background information. Interestingly, I previously was using `conflate()` adjacent to the `advertisements` flow at the same time I saw these ANRs appear (they don't appear immediately, and can take a while to happen). Based on what you said about the binder thread, I am wondering if it would be possible for another process such a WorkManager worker to hang up the process.

I definitely don't need to read every advertisement, since one of the peripherals we are filtering for advertises pretty aggressively at all times. Makes me wonder if I should just do a more periodic scan with `CALLBACK_TYPE_FIRST_MATCH` or something of that sort.

I'll double-check that my `map` function isn't somehow getting stuck or more expensive than I realized.

twyatt, 2024-05-22:
Ya, it is possible to block those binder threads in other areas and affect it in your advertisements collection. If you consume fast enough though, I'd hope it would be fine. 🤷

When using `conflate`, just make sure it is before your expensive flow operation, or as early in the flow operator chain as makes sense for your use case.

Hopefully you have some more things to try now. Let me know though if you still have issues and need further help. Good luck!

djweber, 2024-05-22:
Even with `conflate()`, I still encounter that. Very bizarre.

You mentioned that some vendors can use the main thread as the binder thread - I am wondering if that is what this Samsung Galaxy S10 is doing. I've had the same app running continuously on several different devices, but this one in particular seems to be a lot more ANR-prone. The others haven't encountered a single ANR.

twyatt, 2024-05-22:
`conflate` really _should_ relieve any "back pressure" within the callback, _but_ I can try one last thing...

I'll spin up a `SNAPSHOT` build (likely in the coming days) for you to try where `offer` is used in the callback rather than `trySendBlocking`. I'm skeptical it will help, as coroutines should essentially accomplish the same affect w/ `conflate`, but it is worth a shot to experiment with it.

djweber, 2024-05-22:
I appreciate that - I will keep an eye out.

djweber, 2024-05-23:
Same thing happened with this as well. What's interesting is the app was fine for hours, then when I had to reopen it due to a crash, it ANR'd within a few minutes.

djweber, 2024-05-23:
So far, things have been holding for the past 12 hours with this adjustment: 

https://github.com/djweber/kable/blob/0.31.1-trysend-experiment/core/src/androidMain/kotlin/BluetoothLeScannerAndroidScanner.kt#L40

Assuming that this change actually helps, it almost feels too good to be true 😄 Is there a particular risk to using non-blocking behavior here?

twyatt, 2024-05-23:
Wow, great! That is what I was going to try out for you in the `SNAPSHOT` build.

> Is there a particular risk to using non-blocking behavior here?

There is no risk, you're merely opting into dropping any values that the consumer can't keep up with. The main reason that that isn't the default behavior for Kable is because Kable needed to provide an API where such behavior could be opt-in (i.e. via flow operators, such as `conflate`).

It's unfortunate that `conflate` didn't do the trick. I'll try to find some time soonish to make this behavior configurable (opt-in) so that it doesn't break existing Kable users, but users who need this behavior (such as yourself) can opt-in via a configuration flag of some sort.

djweber, 2024-05-23:
I'm wondering if the `conflate` would help in this specific scenario if something else was making `trySendBlocking` hang upstream besides a slow downstream collection. This is somewhat of a busy application, so certainly possible something else is gumming up the binder and causes a deadlock.

twyatt, 2024-05-23:
@djweber quick question: are you using any `Filter.NamePrefix` for filtering w/ the Scanner? I realized that Kable does have a [`filter` operation on the `Scanner.advertisements` flow](https://github.com/JuulLabs/kable/blob/616488b309aab68be24006ad07461854767779c5/core/src/androidMain/kotlin/BluetoothLeScannerAndroidScanner.kt#L100-L106) that could be applying a little bit of back pressure on the callback flow. It probably isn't an expensive operation, but curious if you happen to have any of those name filters. If so, then it might be worth me benchmarking that `filter` operation to make sure it isn't problematic.

djweber, 2024-05-23:
Not at the moment. The only filters I am applying are a couple of the service UUID filters:

```
filters = listOf(
    Filter.Service(UUID.fromString(...)),
    Filter.Service(UUID.fromString(....))
)
```

djweber, 2024-05-25:
@twyatt Took an initial stab at making that setting configurable: https://github.com/JuulLabs/kable/pull/684/

twyatt, 2024-05-26:
Thanks for the PR! I'll take a look at it in the coming days.

PR #684 (2024-06-03): Provide ability to pre-conflate scanned advertisements
It was observed in #654 that using the default blocking send under certain conditions on various devices could result in ANRs being triggered when performing scans.  This PR provides a way for users to opt in to make the Android scanner's `callbackFlow` use a non-blocking `trySend` instead of `trySendBlocking`, with no change to the existing behavior.

Closes #654
closed 2024-06-03 by twyatt