JuulLabs/kable #662: Services should be re-discovered once onServiceChanged is called

JonatanPlesko, 2024-03-28:
In API 31, [onServiceChanged](https://developer.android.com/reference/android/bluetooth/BluetoothGattCallback#onServiceChanged(android.bluetooth.BluetoothGatt)) was added to BluetoothGattCallback, which Kable currently does not override and handle.

When this callback is called, Kable needs to rediscover services and **turn on notifications/indications** again. I am currently experiencing an issue where none of the previously set observers work as Android silently clears up all of the existing notification/indication registrations when this occurs - see [L1141 in bta_gattc_act.cc](https://android-review.googlesource.com/c/platform/system/bt/+/1433963).

I'm not sure if rediscovering services and re-enabling notifications/indications is sufficient in this scenario for everything to work properly, but I can confirm that manually calling `BluetoothGatt#setCharacteristicNotification` fixes _my_ issue.

As for reproducing the issue, it should occur when the server changes one (or more) of its services while you're already connected to it and subscribed to some characteristics.

twyatt, 2024-03-30:
Thanks for reporting this! I'll try to find time soon to work on it.

The device you're experiencing this issue with, and testing with, does it happen to be a publicly available device? As I don't believe any of the devices I currently have perform a peripheral since service change that would trigger this.

If there isn't a publicly available device to test against, I'll probably end up just calling `setNotification` as you suggested.

JonatanPlesko, 2024-04-01:
Unfortunately, I am not using a publicly available device. It's probably possible to mock such behaviour on Android, but I haven't had the time to try.
BTW, I think calling BluetoothGatt#discoverServices also needs to happen; it's not relevant in my scenario since all of the services and characteristics I was subscribed to are still there, but I think that someone else could run into issues there.

PR #754 (2024-09-26): Handle when peripheral services change
Re-discovers services when peripheral service database changes.

> [!WARNING]
> Does not re-discover on JavaScript, as event is not supported on Chrome, per [Implementation Status](https://github.com/WebBluetoothCG/web-bluetooth/blob/main/implementation-status.md#gatt-communication-api):
> [image]

Closes #662
closed 2024-09-26 by twyatt