JuulLabs/kable #677: Cannot reobserve a characteristic after canceling a job containing a call to observe

mmeisel, 2024-05-16:
I have the same use case as in Q&A answer https://github.com/JuulLabs/kable/discussions/592 except that I sometimes need to re-observe the characteristic during the same connection.

Using kable 0.31.1 on Android, I did exactly as the answer suggested, using `cancelAndJoin()` to cancel a `launch`ed job. This does successfully write `DISABLE_NOTIFICATION_VALUE` to the descriptor. However, when I call `observe()` on the characteristic again, kable seems to think the characteristic is already being observed, and does not write `ENABLE_NOTIFICATION_VALUE`.

I've tracked this down (using the debugger) to the fact that `didStartObservation` is never set to `false` -- I set a breakpoint at [this line](https://github.com/JuulLabs/kable/blob/7a58126d6b61d4ff2b014fb7fb2e834560357374/core/src/commonMain/kotlin/Observation.kt#L76), and it is never reached. AFAICT, this is because the cancellation of the job causes the next `suspend` function, which happens to be [`BluetoothDeviceAndroidPeripheral.setConfigDescriptor()`](https://github.com/JuulLabs/kable/blob/7a58126d6b61d4ff2b014fb7fb2e834560357374/core/src/androidMain/kotlin/BluetoothDeviceAndroidPeripheral.kt#L362), to throw a `CancellationException` on completion, which is propagated back to [the previous line](https://github.com/JuulLabs/kable/blob/7a58126d6b61d4ff2b014fb7fb2e834560357374/core/src/commonMain/kotlin/Observation.kt#L74) in `Observation.kt`.

twyatt, 2024-05-16:
Thanks for the thorough bug report! I'll try to take a look at it soon. PRs are also welcome.

mmeisel, 2024-05-16:
Thanks for the quick response @twyatt! I'd be happy to submit a PR if I could figure out the root cause! I was able to work around this issue by using a different un-observation strategy -- `takeWhile() { stillObserving }` on the returned flow, where `stillObserving` is a volatile boolean. This seems to work fine.

PR #699 (2024-06-27): Ensure teardown is completed when observation coroutines are canceled
It turns out that any suspending function called after cancelation of the coroutine will itself be canceled. [The kotlin documentation](https://kotlinlang.org/docs/cancellation-and-timeouts.html#run-non-cancellable-block) recommends using `withContext(NonCancellable) {}` in these cases.

Fixes #677

I wanted to write a test for this, but the observation tests don't use any suspend functions so it wasn't clear to me how. It did fix the problem for me, however.