JuulLabs/kable #697: iOS Advertisement always null

R4leven, 2024-06-20:
Hi,

We want to filter out non-connectable BLE devices based on some conditions and are using the isConnectable flag on the Advertisement.

The flag works on Android, but on iOS ist is always null, even for connectable devices.

Maybe the cast of the CBAdvertisementDataIsConnectable value to Boolean in is not working correctly.

https://github.com/JuulLabs/kable/blob/aee205e9127853752c06e5d62a87590cbbf3aabf/core/src/appleMain/kotlin/CBPeripheralCoreBluetoothAdvertisement.kt#L43

I tested it without KMP and a CBCentralManager on iOS and the value was set to 1.
Perhaps the value needs to be cast to NSNumber (which is stated here: https://developer.apple.com/documentation/corebluetooth/cbadvertisementdataisconnectable) and then the boolValue needs to be used.

[image]

using com.juul.kable:core:0.31.1 and iOS 17.5.1

twyatt, 2024-06-20:
Aw yes, that looks like a bug. Thanks for the thorough analysis of the issue! I'll get a fix up soon.

twyatt, 2024-06-21:
@R4leven let me know if you need me to back port the fix to Kable 0.32.x (i.e. if you're not ready for Kable 0.33.x w/ K2 upgrade yet).

R4leven, 2024-06-21:
Thank you for the quick reply and fix. We are planning on upgrading to 0.33.x soon, so no back port is needed.

PR #700 (2024-06-24): Cast to `NSNumber` for determining if advertisement is connectable
Closes #697
closed 2024-06-24 by twyatt