JuulLabs/kable #698: .observe not firing any event

leonardovallem, 2024-06-20:
### Problem
characteristic.observe is not firing any events.

Taking a look at the logs, the line `setCharacteristicNotification() - uuid: 00002a05-0000-1000-8000-00805f9b34fb enable: true` from Android appears.
I tested on iOS as well, same behaviour.
I even set some breakpoints for every `onCharacteristicChanged` and `onCharacteristicRead` methods on androidMain > Callback.kt, but the execution didn't reach any of them.

### Kable logs

```
2024-06-20 10:33:45.276  8131-8131  Kable/Peripheral        br.ind.onboard.instalador            I  64:B7:08:CE:90:56 Connecting
2024-06-20 10:33:45.581  8131-8264  Kable/Callback          br.ind.onboard.instalador            D  64:B7:08:CE:90:56 onConnectionStateChange
                                                                                                    status: Success
                                                                                                    newState: Connected
2024-06-20 10:33:45.594  8131-8131  Kable/Peripheral        br.ind.onboard.instalador            V  64:B7:08:CE:90:56 discoverServices
2024-06-20 10:33:45.917  8131-8131  Kable/Scanner           br.ind.onboard.instalador            W  Unable to deliver scan result due to failure in flow or premature closing.
2024-06-20 10:33:46.267  8131-8264  Kable/Callback          br.ind.onboard.instalador            D  64:B7:08:CE:90:56 onServicesDiscovered
                                                                                                    status: GATT_SUCCESS(0)
2024-06-20 10:33:46.272  8131-8131  Kable/Peripheral        br.ind.onboard.instalador            V  64:B7:08:CE:90:56 Discovered 3 services
2024-06-20 10:33:46.272  8131-8131  Kable/Peripheral        br.ind.onboard.instalador            V  64:B7:08:CE:90:56 Configuring characteristic observations
2024-06-20 10:33:46.273  8131-8131  Kable/Peripheral        br.ind.onboard.instalador            I  64:B7:08:CE:90:56 Connected
2024-06-20 10:33:46.279  8131-8131  Kable/Peripheral        br.ind.onboard.instalador            D  64:B7:08:CE:90:56 setCharacteristicNotification
                                                                                                    service: 00001801-0000-1000-8000-00805f9b34fb
                                                                                                    characteristic: 00002a05-0000-1000-8000-00805f9b34fb
                                                                                                    value: true
2024-06-20 10:33:46.283  8131-8131  Kable/Peripheral        br.ind.onboard.instalador            V  64:B7:08:CE:90:56 Writing ENABLE_INDICATION_VALUE to CCCD
                                                                                                    service: 00001801-0000-1000-8000-00805f9b34fb
                                                                                                    characteristic: 00002a05-0000-1000-8000-00805f9b34fb
                                                                                                    descriptor: 00002902-0000-1000-8000-00805f9b34fb
2024-06-20 10:33:46.284  8131-8131  Kable/Peripheral        br.ind.onboard.instalador            D  64:B7:08:CE:90:56 write
                                                                                                    service: 00001801-0000-1000-8000-00805f9b34fb
                                                                                                    characteristic: 00002a05-0000-1000-8000-00805f9b34fb
                                                                                                    descriptor: 00002902-0000-1000-8000-00805f9b34fb
                                                                                                    data: 02 00
2024-06-20 10:33:46.301  8131-8264  Kable/Callback          br.ind.onboard.instalador            D  64:B7:08:CE:90:56 onDescriptorWrite
                                                                                                    service: 00001801-0000-1000-8000-00805f9b34fb
                                                                                                    characteristic: 00002a05-0000-1000-8000-00805f9b34fb
                                                                                                    descriptor: 00002902-0000-1000-8000-00805f9b34fb
                                                                                                    status: GATT_SUCCESS(0)
```

### Code

```kotlin
fun connect() {
    screenModelScope.launch {
        try {
            device.connect()
            wasRestartedByUser = false
            observeLog()
        } catch (e: Exception) {
            _navigationEvent.send(e)
        }
    }
}

private suspend fun observeLog() = device.services
    .orEmpty()
    .flatMap { it.characteristics }
    .forEach {
        it
            .let(device::observe)
            .collect {
                Logger.i("Characteristic read", tag = "Bluetooth")
            }
    }
```

Am I missing something?
Thanks in advance.

twyatt, 2024-06-20:
Based on your code, only the first characteristic in `device.services` will be `observe`d, as `collect` will suspend during the collection and `forEach` won't trigger for any additional characteristics.

Although, for that first characteristic the `observe` should work — assuming it is an observable characteristic on your device?

> I even set some breakpoints for every `onCharacteristicChanged` and `onCharacteristicRead` methods on androidMain

If those callbacks aren't being triggered, then Android isn't informing Kable of any characteristic changes. Are you sure that the BLE peripheral is emitting characteristic changes for this specific characteristic?

leonardovallem, 2024-06-20:
> and forEach won't trigger for any additional characteristics
makes sense, moving the observe call to a launch {} scope for each characteristic did it

thanks for the quick reply and for the help!

twyatt, 2024-06-22:
It sounds like you figured out the issue. Let me know if you have further problems and/or need this issue re-opened.

closed 2024-06-22 by twyatt