JuulLabs/kable #708: Unhelpful exception is thrown when `Filter.Address` is used on Apple platforms

twyatt, 2024-07-05:
```kotlin
val scanner = Scanner {
    filters = listOf(Filter.Address("01:02:03:04:05"))
}
```

| &nbsp; | Exception |
|-|-|
| Expected | `IllegalArgumentException: Filtering by address (Filter.Address) is not supported on Apple platforms` |
| Actual | `FileFailedToInitializeException: There was an error during file or class initialization` |

<details>
<summary>Stacktrace</summary>

```java
> Task :app:runDebugExecutableMacosArm64 FAILED
[I/SensorTag] Searching for SensorTag...
Uncaught Kotlin exception: kotlin.native.internal.FileFailedToInitializeException: There was an error during file or class initialization
    at 0   sensortag.kexe                      0x10014d12f        kfun:kotlin.Error#<init>(kotlin.String?;kotlin.Throwable?){} + 143 (/opt/buildAgent/work/b2e1db4d8d903ca4/kotlin/kotlin-native/runtime/src/main/kotlin/kotlin/Exceptions.kt:14:70)
    at 1   sensortag.kexe                      0x100171a0b        kfun:kotlin.native.internal.FileFailedToInitializeException#<init>(kotlin.String?;kotlin.Throwable?){} + 143 (/opt/buildAgent/work/b2e1db4d8d903ca4/kotlin/kotlin-native/runtime/src/main/kotlin/kotlin/native/internal/RuntimeUtils.kt:146:87)
    at 2   sensortag.kexe                      0x100170c67        ThrowFileFailedToInitializeException + 275 (/opt/buildAgent/work/b2e1db4d8d903ca4/kotlin/kotlin-native/runtime/src/main/kotlin/kotlin/native/internal/RuntimeUtils.kt:158:15)
    at 3   sensortag.kexe                      0x1002342d7        CallInitGlobalPossiblyLock + 759
    at 4   sensortag.kexe                      0x1000dcfe7        kfun:com.juul.sensortag#<get-scanner>(){}com.juul.kable.Scanner<com.juul.kable.PlatformAdvertisement> + 75 (/Users/travis/Projects/sensortag/app/src/commonMain/kotlin/SensorTag.kt:42:1)
    at 5   sensortag.kexe                      0x1000db35b        kfun:com.juul.sensortag.$main$lambda$16COROUTINE$2.invokeSuspend#internal + 787 (/Users/travis/Projects/sensortag/app/src/appleMain/kotlin/main.kt:21:25)
    at 6   sensortag.kexe                      0x1000db837        kfun:com.juul.sensortag.main$lambda$16#internal + 251 (/Users/travis/Projects/sensortag/app/src/appleMain/kotlin/main.kt:16:32)
    at 7   sensortag.kexe                      0x1000db8ef        kfun:com.juul.sensortag.$main$lambda$16$FUNCTION_REFERENCE$0.invoke#internal + 115 (/Users/travis/Projects/sensortag/app/src/appleMain/kotlin/main.kt:16:14)
    at 8   sensortag.kexe                      0x1001ce9d7        kfun:kotlin.Function2#invoke(1:0;1:1){}1:2-trampoline + 115 (/Users/teamcity/.gradle/daemon/8.4/[K][Suspend]Functions:1:1)
    at 9   sensortag.kexe                      0x10015a3cb        kfun:kotlin.coroutines.intrinsics.object-4.invokeSuspend#internal + 731 (/opt/buildAgent/work/b2e1db4d8d903ca4/kotlin/kotlin-native/runtime/src/main/kotlin/kotlin/coroutines/intrinsics/IntrinsicsNative.kt:254:35)
    at 10  sensortag.kexe                      0x1001ce2f3        kfun:kotlin.coroutines.native.internal.BaseContinuationImpl#invokeSuspend(kotlin.Result<kotlin.Any?>){}kotlin.Any?-trampoline + 67 (/opt/buildAgent/work/b2e1db4d8d903ca4/kotlin/kotlin-native/runtime/src/main/kotlin/kotlin/coroutines/ContinuationImpl.kt:50:24)
    at 11  sensortag.kexe                      0x1001579b3        kfun:kotlin.coroutines.native.internal.BaseContinuationImpl#resumeWith(kotlin.Result<kotlin.Any?>){} + 623 (/opt/buildAgent/work/b2e1db4d8d903ca4/kotlin/kotlin-native/runtime/src/main/kotlin/kotlin/coroutines/ContinuationImpl.kt:30:39)
    at 12  sensortag.kexe                      0x1001ce3d3        kfun:kotlin.coroutines.Continuation#resumeWith(kotlin.Result<1:0>){}-trampoline + 99 (/opt/buildAgent/work/b2e1db4d8d903ca4/kotlin/libraries/stdlib/src/kotlin/coroutines/Continuation.kt:26:12)
    at 13  sensortag.kexe                      0x1002aa35b        kfun:kotlinx.coroutines.DispatchedTask#run(){} + 1879 (/opt/buildAgent/work/44ec6e850d5c63f0/kotlinx-coroutines-core/common/src/internal/DispatchedTask.kt:104:71)
    at 14  sensortag.kexe                      0x1002cf93b        kfun:kotlinx.coroutines.Runnable#run(){}-trampoline + 91 (/opt/buildAgent/work/44ec6e850d5c63f0/kotlinx-coroutines-core/native/src/Runnable.kt:10:19)
    at 15  sensortag.kexe                      0x100250393        kfun:kotlinx.coroutines.EventLoopImplBase#processNextEvent(){}kotlin.Long + 1247 (/opt/buildAgent/work/44ec6e850d5c63f0/kotlinx-coroutines-core/common/src/EventLoop.common.kt:277:44)
    at 16  sensortag.kexe                      0x1002cf687        kfun:kotlinx.coroutines.EventLoop#processNextEvent(){}kotlin.Long-trampoline + 51 (/opt/buildAgent/work/44ec6e850d5c63f0/kotlinx-coroutines-core/common/src/EventLoop.common.kt:49:10)
    at 17  sensortag.kexe                      0x1002c7827        kfun:kotlinx.coroutines.BlockingCoroutine.joinBlocking#internal + 435 (/opt/buildAgent/work/44ec6e850d5c63f0/kotlinx-coroutines-core/native/src/Builders.kt:129:43)
    at 18  sensortag.kexe                      0x1002c668b        kfun:kotlinx.coroutines#runBlocking(kotlin.coroutines.CoroutineContext;kotlin.coroutines.SuspendFunction1<kotlinx.coroutines.CoroutineScope,0:0>){0§<kotlin.Any?>}0:0 + 1395 (/opt/buildAgent/work/44ec6e850d5c63f0/kotlinx-coroutines-core/native/src/Builders.kt:68:26)
    at 19  sensortag.kexe                      0x1002c686b        kfun:kotlinx.coroutines#runBlocking$default(kotlin.coroutines.CoroutineContext?;kotlin.coroutines.SuspendFunction1<kotlinx.coroutines.CoroutineScope,0:0>;kotlin.Int){0§<kotlin.Any?>}0:0 + 239 (/opt/buildAgent/work/44ec6e850d5c63f0/kotlinx-coroutines-core/native/src/Builders.kt:45:15)
    at 20  sensortag.kexe                      0x1000d96f7        kfun:com.juul.sensortag#main(){} + 107 (/Users/travis/Projects/sensortag/app/src/appleMain/kotlin/main.kt:16:14)
    at 21  sensortag.kexe                      0x1000d978b        Konan_start + 111 (/Users/travis/Projects/sensortag/app/src/appleMain/kotlin/main.kt:16:1)
    at 22  sensortag.kexe                      0x1000e6b43        Init_and_run_start + 107
    at 23  dyld                                0x192e5a0df        start + 2359
Caused by: kotlin.IllegalArgumentException: Filtering by address (`Filter.Address`) is not supported on Apple platforms
    at 0   sensortag.kexe                      0x1001530e7        kfun:kotlin.Throwable#<init>(kotlin.String?){} + 119 (/opt/buildAgent/work/b2e1db4d8d903ca4/kotlin/kotlin-native/runtime/src/main/kotlin/kotlin/Throwable.kt:28:44)
    at 1   sensortag.kexe                      0x10014d223        kfun:kotlin.Exception#<init>(kotlin.String?){} + 115 (/opt/buildAgent/work/b2e1db4d8d903ca4/kotlin/kotlin-native/runtime/src/main/kotlin/kotlin/Exceptions.kt:23:51)
    at 2   sensortag.kexe                      0x10014d3bb        kfun:kotlin.RuntimeException#<init>(kotlin.String?){} + 115 (/opt/buildAgent/work/b2e1db4d8d903ca4/kotlin/kotlin-native/runtime/src/main/kotlin/kotlin/Exceptions.kt:34:51)
    at 3   sensortag.kexe                      0x10014d6b3        kfun:kotlin.IllegalArgumentException#<init>(kotlin.String?){} + 115 (/opt/buildAgent/work/b2e1db4d8d903ca4/kotlin/kotlin-native/runtime/src/main/kotlin/kotlin/Exceptions.kt:59:51)
    at 4   sensortag.kexe                      0x1000a6c6f        kfun:com.juul.kable.CentralManagerCoreBluetoothScanner#<init>(com.juul.kable.CentralManager;kotlin.collections.List<com.juul.kable.Filter>;kotlin.collections.Map<kotlin.Any?,*>?;com.juul.kable.logs.Logging){} + 963 (/Users/travis/Projects/kable/core/src/appleMain/kotlin/CentralManagerCoreBluetoothScanner.kt:19:50)
    at 5   sensortag.kexe                      0x1000bf237        kfun:com.juul.kable.ScannerBuilder#build(){}com.juul.kable.Scanner<com.juul.kable.PlatformAdvertisement> + 1639 (/Users/travis/Projects/kable/core/src/appleMain/kotlin/ScannerBuilder.kt:38:100)
    at 6   sensortag.kexe                      0x1000d0e8f        kfun:com.juul.kable#Scanner(kotlin.Function1<com.juul.kable.ScannerBuilder,kotlin.Unit>){}com.juul.kable.Scanner<com.juul.kable.PlatformAdvertisement> + 227 (/Users/travis/Projects/kable/core/src/commonMain/kotlin/Scanner.kt:12:60)
    at 7   sensortag.kexe                      0x1000dc90b        kfun:com.juul.sensortag.$init_global#internal + 883 (/Users/travis/Projects/sensortag/app/src/commonMain/kotlin/SensorTag.kt:1:1)
    at 8   sensortag.kexe                      0x1002341fb        CallInitGlobalPossiblyLock + 539
    at 9   sensortag.kexe                      0x1000dcfe7        kfun:com.juul.sensortag#<get-scanner>(){}com.juul.kable.Scanner<com.juul.kable.PlatformAdvertisement> + 75 (/Users/travis/Projects/sensortag/app/src/commonMain/kotlin/SensorTag.kt:42:1)
    ... and 19 more common stack frames skipped
```
</details>

This is caused by throwing the exception in the `CentralManagerCoreBluetoothScanner` class' init block:

https://github.com/JuulLabs/kable/blob/c91f0fa9bdb5469b55ecc29aad31ce5760f24d21/core/src/appleMain/kotlin/CentralManagerCoreBluetoothScanner.kt#L26-L30

dmikots, 2024-08-07:
@twyatt Hi. Can u add an iOS Filter by mac address from advertisement data?

twyatt, 2024-08-07:
Unfortunately (as far as I'm aware), Core Bluetooth (iOS) does not expose the MAC address, making it impossible to filter by that on iOS.

Let me know if you happen to find a Core Bluetooth API that exposes MAC address.