JuulLabs/kable #714: Multiple ProfileServiceNotBound errors when trying to write to a Peripheral

PavlosTze, 2024-07-16:
Hello,

I have the following code:
```
    private suspend fun enableDescriptorNotification(descriptor: Descriptor) {
        try {
            peripheral.write(descriptor, ENABLE_NOTIFICATION_VALUE)
        } catch (e: GattWriteException) {
            Timber.w(e, "[enable descriptor notification]: GattWriteException - Result: ${e.result}")
        }
    }
```

And for some weeks already this code is producing **a lot** of `ProfileServiceNotBound` types of exception, the log is the following:
> Non-fatal Exception: f7.N: Write failed: ProfileServiceNotBound
> at com.juul.kable.BluetoothGattKt.writeCharacteristicOrThrow(BluetoothGatt.kt:58)
> at com.juul.kable.BluetoothDeviceAndroidPeripheral$write$$inlined$execute$1.invokeSuspend(Connection.kt:126)
> at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
> at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:104)
> at android.os.Handler.handleCallback(Handler.java:959)
> at android.os.Handler.dispatchMessage(Handler.java:100)
> at android.os.Looper.loopOnce(Looper.java:232)
> at android.os.Looper.loop(Looper.java:317)
> at android.os.HandlerThread.run(HandlerThread.java:85)

As you can see from these 3 screenshots issues started in early June:
[image]
[image]
[image]

In case that helps, our app producing those is [Open Source](https://github.com/WeatherXM/wxm-android/blob/2730edea4f2ef27e4f00ebcfacec763d9731c15d/app/src/main/java/com/weatherxm/data/bluetooth/BluetoothConnectionManager.kt#L234).

Based on what I read in your [writeResultFrom](https://github.com/JuulLabs/kable/blob/c9fc138af19c3023f4906be6ee3065d6d0156eaf/core/src/androidMain/kotlin/BluetoothGatt.kt#L94) and in the [Android-related code](https://cs.android.com/android/platform/superproject/main/+/b7a389a145ff443550e1a942bf713c60c2bd6a14:packages/modules/Bluetooth/framework/java/android/bluetooth/BluetoothStatusCodes.java;l=70):
```
    /**
     * Error code indicating that the profile service is not bound. You can bind a profile service
     * by calling {@link BluetoothAdapter#getProfileProxy}.
     */
    public static final int ERROR_PROFILE_SERVICE_NOT_BOUND = 9;
```

Any chance you can catch this error without letting it fail? What is this related to? As there aren't any logs producing it in the past and we've been using Kable for around 2 years already.

twyatt, 2024-07-16:
Can you provide a breakdown of the failure rate per Android version?
I'm wondering if this is the result of behavior changes in newer versions of Android?

PavlosTze, 2024-07-17:
> Can you provide a breakdown of the failure rate per Android version? I'm wondering if this is the result of behavior changes in newer versions of Android?

Sure, here it is:
[image]

twyatt, 2024-07-18:
Thanks for the info. We'll seeing this on our apps as well. I'll try to allocate some time soon to investigate.

twyatt, 2024-07-23:
@PavlosTze I'm currently investigating this issue. Do you happen to know what version of Kable you first saw this issue in?

PavlosTze, 2024-07-24:
@twyatt It started with 0.31.1.

twyatt, 2024-07-24:
Thanks, I believe I've tracked down the cause:

I believe it was introduced in #648, when a bug was fixed (whereas Kable would silently ignore failures to write to a characteristic). In #648, that behavior was fixed (to now throw an exception if the write failed).

Unfortunately, some might be relying on the bug where it didn't propagate a failure when the write fails.

I'm trying to determine what the best API might be for this situation. I'll follow up soon.

PavlosTze, 2024-07-24:
Sure, I just want to mention that this is a **non-crash failure**. Which means that if it failed before in the same way, the only thing happening now is that it's been showing up in the logs, not that big of a deal.

I'm wondering if we can solve it in the background based on what it says [here](https://cs.android.com/android/platform/superproject/main/+/b7a389a145ff443550e1a942bf713c60c2bd6a14:packages/modules/Bluetooth/framework/java/android/bluetooth/BluetoothStatusCodes.java;l=70)?
```
    /**
     * Error code indicating that the profile service is not bound. You can bind a profile service
     * by calling {@link BluetoothAdapter#getProfileProxy}.
     */
    public static final int ERROR_PROFILE_SERVICE_NOT_BOUND = 9;
```
Is it possible to bind a profile service if we get across this failure?

twyatt, 2024-07-24:
I suspect that the comment you referenced is misleading in this case.

I believe the failure happens when a write occurs after a connection is lost / disposed; because the connection was disposed, the underlying Android GATT object no longer has a "profile service" (essentially a reference to the underlying Android BLE system) and so propagates that failure. I believe it is unrelated to `getProfileProxy`; perhaps Android just re-used this failure case for multiple failure conditions. 🤷 

I think in most cases it is safe to ignore the failure (in the cases where it happens because the connection was recently disposed). The case where it might be problematic is where `BluetoothAdapter#getProfileProxy` is being utilized (I suspect this isn't common though).

twyatt, 2024-11-06:
The connection management, especially around I/O has been refactored. Can you try [0.35.0-rc](https://github.com/JuulLabs/kable/releases/tag/0.35.0-rc) and see if the issue has been resolved?

PavlosTze, 2024-11-09:
Hey, unfortunately I couldn't reproduce the error, but there are numerous crashlytics logs from our users. But...As I saw the changes of `0.35.0-rc` and there are plenty of them, I can try to find some time in the next few days to test it myself and come back with any issues I may found (or just communicate that everything looked OK). How does that sound?

twyatt, 2024-11-10:
That sounds great. I appreciate you taking the time to help test. ❤️

PavlosTze, 2024-11-12:
> That sounds great. I appreciate you taking the time to help test. ❤️

Hey, I tested it thoroughly today, I opened some issues. FYI our project is open-source, in case you want to have a look at how we use Kable, you can view our `BluetoothConnectionManager`:
- [Version supporting 0.35.0-RC](https://github.com/WeatherXM/wxm-android/blob/feature/WIP-TEST-BLE-Library-Kable-0.35.0-rc/app/src/main/java/com/weatherxm/data/bluetooth/BluetoothConnectionManager.kt)
- [Version up to 0.34.0](https://github.com/WeatherXM/wxm-android/blob/main/app/src/main/java/com/weatherxm/data/bluetooth/BluetoothConnectionManager.kt)

Feel free to ask me if anything, happy to help.

twyatt, 2024-11-12:
Thanks for reporting all those issues! Super helpful!

twyatt, 2024-12-06:
Removed milestone, as I believe Kable `0.35.x` releases provide a means to properly shutdown tasks with a connection, and should provide a means to avoid these failures.

Leaving open for a while for further feedback/validation.