JuulLabs/kable #729: Feature request: Add maximumWriteValueLengthForType to iOS Peripheral

NaiRo94s, 2024-08-01:
Hi, 
it would be great if you could add [maximumWriteValueLengthForType](https://developer.apple.com/documentation/corebluetooth/cbperipheral/maximumwritevaluelength(for:)) to the CoreBluetoothPeripheral interface.

Sample: 

```
    override fun maximumWriteValueLengthForType(writeType: WriteType): Int {
        val type = when (writeType) {
            WithResponse -> CBCharacteristicWriteWithResponse
            WithoutResponse -> CBCharacteristicWriteWithoutResponse
        }
        return cbPeripheral.maximumWriteValueLengthForType(type).toInt()
    }
```

twyatt, 2024-08-01:
Thanks for the code snippet. This would be a good addition. I'll try to add it soon.

matttbates, 2024-11-20:
Hey Travis, I have a use-case that requires this functionality. Are there any plans to add this in the near future? Thank you for your support and for a solid KMP library.

twyatt, 2024-11-21:
I would like to get to this sooner rather than later, but not sure I'll have much bandwidth in the near term.
I may be able to find some time after getting a final `0.35.0` release out.

twyatt, 2024-11-21:
PRs are also welcome, and I try to prioritize them above most other work I have on my plate. 😉

ErikHellman, 2024-12-03:
I made a PR implementing this. I haven't been able to test this properly, but as I'm only reading data I believe this should be fairly safe in any case.

PR #809 (2024-12-06): Adds maximumWriteValueLength to Peripheral
- Calls maximumWriteValueLength on iOS
- Returns latest know value from onMtuChanged() - 3 on Android, or the default (23 - 3) if the callback haven't been called
- Return default (23 -3) on Web

Closes #729
closed 2024-12-06 by twyatt