JuulLabs/kable #730: bug: Duplicate class ExceptionsKt

EinfachHans, 2024-08-05:
## Description

Hey there,

i just came across this library while integrating an .aar that depends on this. According to their docs i should add

```gradle
implementation('com.juul.kable:core:0.32.0')
```

With this i get this error on build:

```
Duplicate class com.juul.kable.ExceptionsKt found in modules core-jvm (com.juul.kable:core-jvm:0.32.0) and exceptions-jvm (com.juul.kable:exceptions-jvm:0.32.0)
```

Can you help me out with this?

twyatt, 2024-08-06:
I'm assuming you're integrating this into an Android app?

It isn't clear to me yet why you're hitting this but I haven't yet. Can you provide the output of:

```
./gradlew :app:dependencyInsight --configuration debugCompileClasspath --dependency com.juul.kable
```

It might help shed some light on what is going wrong.

EinfachHans, 2024-08-06:
Hey @twyatt 😊 

Yes we are talking about an android app, created with capacitor.

<details>
<summary>Output of your mentioned command</summary>

```
Starting a Gradle Daemon (subsequent builds will be faster)

> Configure project :app
WARNING: Using flatDir should be avoided because it doesn't support any meta-data formats.

> Configure project :capacitor-cordova-android-plugins
WARNING: Using flatDir should be avoided because it doesn't support any meta-data formats.

> Task :app:dependencyInsight
com.juul.kable:core:0.32.0
  Variant jvmApiElements-published:
    | Attribute Name                                  | Provided | Requested |
    |-------------------------------------------------|----------|-----------|
    | org.gradle.libraryelements                      | jar      |           |
    | org.gradle.status                               | release  |           |
    | org.jetbrains.kotlin.platform.type              | jvm      |           |
    | org.gradle.category                             | library  | library   |
    | org.gradle.usage                                | java-api | java-api  |
    | com.android.build.api.attributes.AgpVersionAttr |          | 8.2.1     |
    | com.android.build.api.attributes.BuildTypeAttr  |          | debug     |
    | org.gradle.jvm.environment                      |          | android   |
   Selection reasons:
      - By constraint: debugRuntimeClasspath uses version 0.32.0

com.juul.kable:core:0.32.0
\--- debugCompileClasspath

com.juul.kable:core:{strictly 0.32.0} -> 0.32.0
\--- debugCompileClasspath

com.juul.kable:core-jvm:0.32.0
  Variant jvmApiElements-published:
    | Attribute Name                                  | Provided | Requested |
    |-------------------------------------------------|----------|-----------|
    | org.gradle.libraryelements                      | jar      |           |
    | org.gradle.status                               | release  |           |
    | org.jetbrains.kotlin.platform.type              | jvm      |           |
    | org.gradle.category                             | library  | library   |
    | org.gradle.usage                                | java-api | java-api  |
    | com.android.build.api.attributes.AgpVersionAttr |          | 8.2.1     |
    | com.android.build.api.attributes.BuildTypeAttr  |          | debug     |
    | org.gradle.jvm.environment                      |          | android   |
   Selection reasons:
      - By constraint: debugRuntimeClasspath uses version 0.32.0
      - By ancestor

com.juul.kable:core-jvm:{strictly 0.32.0} -> 0.32.0
\--- debugCompileClasspath

com.juul.kable:core-jvm:0.32.0
\--- com.juul.kable:core:0.32.0
     \--- debugCompileClasspath

com.juul.kable:exceptions:0.32.0
  Variant jvmApiElements-published:
    | Attribute Name                                  | Provided | Requested |
    |-------------------------------------------------|----------|-----------|
    | org.gradle.libraryelements                      | jar      |           |
    | org.gradle.status                               | release  |           |
    | org.jetbrains.kotlin.platform.type              | jvm      |           |
    | org.gradle.category                             | library  | library   |
    | org.gradle.usage                                | java-api | java-api  |
    | com.android.build.api.attributes.AgpVersionAttr |          | 8.2.1     |
    | com.android.build.api.attributes.BuildTypeAttr  |          | debug     |
    | org.gradle.jvm.environment                      |          | android   |
   Selection reasons:
      - By constraint: debugRuntimeClasspath uses version 0.32.0
      - By ancestor

com.juul.kable:exceptions:{strictly 0.32.0} -> 0.32.0
\--- debugCompileClasspath

com.juul.kable:exceptions:0.32.0
\--- com.juul.kable:core-jvm:0.32.0
     +--- debugCompileClasspath (requested com.juul.kable:core-jvm:{strictly 0.32.0})
     \--- com.juul.kable:core:0.32.0
          \--- debugCompileClasspath

com.juul.kable:exceptions-jvm:0.32.0
  Variant jvmApiElements-published:
    | Attribute Name                                  | Provided | Requested |
    |-------------------------------------------------|----------|-----------|
    | org.gradle.libraryelements                      | jar      |           |
    | org.gradle.status                               | release  |           |
    | org.jetbrains.kotlin.platform.type              | jvm      |           |
    | org.gradle.category                             | library  | library   |
    | org.gradle.usage                                | java-api | java-api  |
    | com.android.build.api.attributes.AgpVersionAttr |          | 8.2.1     |
    | com.android.build.api.attributes.BuildTypeAttr  |          | debug     |
    | org.gradle.jvm.environment                      |          | android   |
   Selection reasons:
      - By constraint: debugRuntimeClasspath uses version 0.32.0
      - By ancestor

com.juul.kable:exceptions-jvm:{strictly 0.32.0} -> 0.32.0
\--- debugCompileClasspath

com.juul.kable:exceptions-jvm:0.32.0
\--- com.juul.kable:exceptions:0.32.0
     +--- debugCompileClasspath (requested com.juul.kable:exceptions:{strictly 0.32.0})
     \--- com.juul.kable:core-jvm:0.32.0
          +--- debugCompileClasspath (requested com.juul.kable:core-jvm:{strictly 0.32.0})
          \--- com.juul.kable:core:0.32.0
               \--- debugCompileClasspath

(*) - Indicates repeated occurrences of a transitive dependency subtree. Gradle expands transitive dependency subtrees only once per project; repeat occurrences only display the root of the subtree, followed by this annotation.

A web-based, searchable dependency report is available by adding the --scan option.

BUILD SUCCESSFUL in 20s
1 actionable task: 1 executed
```
</details>

twyatt, 2024-08-06:
Not sure if it is the cause of your duplicate class build failure, but it looks like your Gradle configuration is pulling in the JVM (no-op) version of Kable (`core-jvm`) for your Android app, when the Android variant should be used (`core-android`).

What version of Gradle (wrapper) are you using w/ your project?

EinfachHans, 2024-08-07:
Hey @twyatt ,

just had a look. Seems to be: **8.2.1**

[image]

twyatt, 2024-08-07:
That seems like a current enough Gradle version (I'd be surprised if it was causing the issue).

At this point, I suspect it is your project configuration (or the configuration of the dependency you mentioned) that is pulling in the JVM version of Kable (instead of the Android version, which you actually need).

You should try and figure out why the `core-jvm` artifact is being used. If you can share your project's Gradle setup, then I can take a look and see if I see anything that looks problematic.

EinfachHans, 2024-08-14:
@twyatt sorry for the late response, i was on vacation 😃 

I'm not an native android developer, can you specify what you mean with "my gradle setup"?

twyatt, 2024-08-20:
> I'm not an native android developer, can you specify what you mean with "my gradle setup"?

Ideally, any Gradle configuration file snippets that you can share that might shed light as to why you're pulling in the JVM version of Kable (on Android), when you should be using the `-android` version.

The output of `./gradlew :app:dependencies` might provide further insight. 🤞

twyatt, 2024-08-21:
Thanks for the small repro!

Unfortunately, I wasn't able to run Gradle diagnostics on it (e.g. `./gradlew :app:dependencyInsight --configuration debugCompileClasspath --dependency com.juul.kable`):

```
A problem occurred evaluating script.
> Could not read script '/Users/travis/Projects/capacitor-kable-bug/android/capacitor-cordova-android-plugins/cordova.variables.gradle' as it does not exist.
```

Could you try changing [`android/app/build.gradle#L44`](https://github.com/EinfachHans/capacitor-kable-bug/blob/4dea62d02a8f159229cb72000ac9ade5a6eaa492/android/app/build.gradle#L44) to read:

```groovy
implementation('com.juul.kable:core-android:0.32.0')
debugImplementation('com.juul.kable:core-android-debug:0.32.0')
```

I would've thought that Gradle would choose the Android variant of the dependencies, but it seems it is not in this case. 🤷 

If that doesn't work, let me know and I'll dig deeper into your sample.

EinfachHans, 2024-08-21:
Did you run steps `2` and `3`? This should create this file 🤔 On the project i switched to `implementation('com.juul.kable:core-android:0.32.0')` which seems to work.

twyatt, 2024-08-21:
Oh. My bad. I jumped straight to using Gradle, since that's what I'm used to. Should've followed your instructions better. 🤦

Glad it sounds so like you got it working though.

Kable will likely drop the extra exception module (that is causing you trouble) soon. Hopefully in an upcoming version this workaround won't be necessary.

closed 2024-08-22 by twyatt