JuulLabs/kable #731: How to identify unexpected disconnects.

Daniel-Sennewald, 2024-08-07:
Hi everyone.

First of all: Thanks for the amazing work on this library. I dreaded BLE on Android and this really made it a breeze.

As the title suggests, I am struggling with identifying unexpected disconnects. The documentation does not provide information on this topic as far as I can see. I also checked issues related to Disconnects and looked into the SensorTag Example. I also did not find any example / information.

I would like to be able to differentiate between disconnects happening due to actively calling the disconnect() method and disconnects happening due to other reasons, e.g. when the peripheral is shutting down.

Initially I was expecting that the disconnect() method would set the State.Disconnected().status Flag. However, when reading peripheral.state I always get Disconnected(null). 

Therefore, I would highly appreciate any hints if this is possible and how it should be done.

twyatt, 2024-08-07:
Is this on Android?

If so, Kable should populate the `Disconnected` reason (assuming provided):

https://github.com/JuulLabs/kable/blob/50157a49363fb81b8eeecbd4856ead353ff7d086/kable-core/src/androidMain/kotlin/gatt/Callback.kt#L112

https://github.com/JuulLabs/kable/blob/50157a49363fb81b8eeecbd4856ead353ff7d086/kable-core/src/androidMain/kotlin/gatt/Callback.kt#L295-L306

You can enable [Kable logging](https://github.com/JuulLabs/kable?tab=readme-ov-file#logging) to get a view of the raw GATT status value that Android provided with the disconnected event.

Let me know if the GATT status that Android provided was something other than `GATT_SUCCESS` (`0`, which Kable maps to `null` on `Disconnected` state).

Daniel-Sennewald, 2024-08-15:
Hi there. Thanks for your timely response.

So for me regardless if I disconnect via the disconnect() method or by just shutting the device down, I get a null on disconnect state.

So this is what the GATT service logs when just shutting the device down:
`11:01:33.308 BluetoothGatt            D  onClientConnectionState() - status=0 clientIf=16 device=deviceid
11:01:33.324                                       D  close()
11:01:33.334                                       D  unregisterApp() - mClientIf=16
11:01:33.375 Peripheral...nlined$map  D  Peripheral connection state changed: Disconnected(null)`

These are the logs when the user disconnects from the device via my UI, i.e. calling the peripheral.disconnect() method.
`11:04:44.579 BluetoothGatt            D  cancelOpen() - device: 74:4D:BD:77:0C:8D
11:04:44.583                                      D  onClientConnectionState() - status=0 clientIf=18 device=74:4D:BD:77:0C:8D
11:04:44.584                                      D  close()
11:04:44.586                                      D  unregisterApp() - mClientIf=18
11:04:44.588 Peripheral...nlined$map  D  Peripheral connection state changed: Disconnected(null)`

The log statement: `11:04:44.588 Peripheral...nlined$map  D  Peripheral connection state changed: Disconnected(null)`
is produced on any state change of the peripheral and simply logs the state. Thus, it should log any of those disconnectedConnectionStatus values you posted. However, in both cases the status is "null".

Ideally, I would like to be able e.g. call peripheral.disconnect(CONNECTION_TERMINATED_BY_USER) to distinguish this user intended case from all the other cases where the connection was disrupted unintentionally.

Anyways, I was able to work around it by updating a "expected disconnect" flag in conjunction with the disconnect() method.
Thanks for your help :)

twyatt, 2024-08-15:
Unfortunately, the logs you shared don't have Kable logs.

It sounds like you've gotten things resolved, but if you need further assistance, please enable Kable logging and share additional logs.

closed 2024-08-15 by twyatt