JuulLabs/kable #737: Remove `Bluetooth.availability`

twyatt, 2024-08-14:
The `Bluetooth.availability` is problematic because it provides `Available` or `Unavailable` but it is inconsistent across platforms.

This comes down to the major differences in how Android vs. Core Bluetooth (e.g. iOS) report their bluetooth radio state and bluetooth permissions.

### Core Bluetooth

Core Bluetooth combines BLE supported status, radio state and permissions via [`CBManagerState`](https://developer.apple.com/documentation/corebluetooth/cbmanagerstate):

- `poweredOff` _(radio state)_
- `poweredOn` _(radio state)_
- `resetting`
- `unauthorized` _(permission)_
- `unknown`
- `unsupported` _(supported status)_

### Android

Android separates BLE supported status, radio state and permissions.

BLE supported status can be checked via the existence (non-`null`) value for the `BluetoothAdapter`. Radio state is provided via `BluetoothAdapter` states:

- `STATE_OFF`
- `STATE_ON`
- `STATE_TURNING_OFF`
- `STATE_TURNING_ON`

Permissions are determined via the Android permissions API.

### Differences

On Android, there is no way of getting real-time permission status (it is checked on-demand) while having permission on iOS comes in via the `CBCentralManager` delegate (a non-`unsupported`) state.

Requesting permission on Android is explicit, while permission request is presented to the user on construction of `CBCentralManager`.

### Deprecation of `Bluetooth.availability`

Because of the major platform differences, there is no way to provide a consistent _flow_ of when BLE is "ready to use" (i.e. available). As such, detecting the availability of BLE is better left to the consumer.

A possible approach will be provided in the SensorTag sample app.

twyatt, 2024-09-19:
**UPDATE:** Even providing `isSupported` has proven challenging (due to Apple limitations with being able to query and **always** showing the user an intrusive permission dialog). Providing `isSupported` in Kable is still being considered.

curioustechizen, 2025-07-24:
I understand why this was closed. I'm wondering if it makes sense for Kable to somehow expose the underlying `CBCentralManager` though. Because unlike Android where BluetoothManager is a singleton, on iOS you can have multiple `CBCentralManager` instances.

It is unclear to me what the implications are of instantiating multiple of them. For example if I wanted to handle Bluetooth state availability on my own, I could simply write `CBCentralManager().state`. Is this valid? Or does this interfere with for example the `unauthorized` state (as in, does CoreBluetooth associate authorization status with a particular instance of `CBCentralManager`)?

twyatt, 2025-07-24:
To be perfectly honest (disclaimer), much of the iOS side of things remains a mystery because Core Bluetooth is closed source and the documentation is often severely lacking. Things are often just determined via trial-and-error.

Specifically w/ regard to instantiating your own `CBCentralManager`: it has thus far been my experience that that should be fine (and as you discovered, that's what I did in SensorTag).

I'm happy to evaluate if exposing the instance that Kable uses is necessary, but my understanding at this point is that it shouldn't be necessary. 🤷

curioustechizen, 2025-08-20:
I will try to put together an MCVE. Before that, I want to clarify a couple of things. Maybe it helps you if I start with my motivation for this feature

## Ensuring Bluetooth is ON before connecting to a peripheral
It looks to me like this is already possible with Kable.
1. `scanner.advertisements` flow throws `UnmetRequirementException` with reason = `BluetoothDisabled` if you try to scan when Bluetooth is OFF. This can be used to nudge the user to turn on Bluetooth.
2. However, not all connections go through a scan. For example, you can get a `Peripheral` from a saved Bluetooth address and you can call `connect()` on it. The API docs for `connect()` don't mention `BluetoothDisabled` but I see at least [iOS](https://github.com/JuulLabs/kable/blob/3daaedb983bdb9ba62cbf5ecf0ffe413a42f71af/kable-core/src/appleMain/kotlin/CBPeripheralCoreBluetoothPeripheral.kt#L112) and [Android](https://github.com/JuulLabs/kable/blob/3daaedb983bdb9ba62cbf5ecf0ffe413a42f71af/kable-core/src/appleMain/kotlin/CBPeripheralCoreBluetoothPeripheral.kt#L112) potentially throw the same `UnmetRequirementException` with reason = `BluetoothDisabled` while trying to establish a connection.

I'm assuming that existing Kable facilities should be sufficient if the use case is to inform the user to turn on Bluetooth when attempting to connect to a peripheral

## Changing the UI based on Bluetooth being powered ON
This seems more challenging with the deprecation of `Bluetooth.availability`. On Android it is fairly trivial. On iOS, it should have been trivial but I see some flakiness around the timing, which I'm guessing is because of instantiation of a CBCentralManager to check the bluetooth state.

----

Do you think these use-cases are valid? Is there some way using Kable (or without) to achieve these use-cases that I'm missing?