JuulLabs/kable #764: reconnect mechanism not working

MohamedEbrahim13, 2024-09-27:
val peripheral = scope.peripheral(..) {
    autoConnectIf { true }
}
peripheral.connect()

twyatt, 2024-09-27:
The code snippet you provided will initiate a connection, but your code doesn't handle **re**connection.

`autoConnectIf` determines if [`autoConnect`](https://developer.android.com/reference/android/bluetooth/BluetoothDevice#connectGatt(android.content.Context,%20boolean,%20android.bluetooth.BluetoothGattCallback,%20int)) should be `true` for Android's `connect` call.

Per Kable's [documentation](https://github.com/JuulLabs/kable?tab=readme-ov-file#autoconnect), `autoConnectIf` configures the connect process, not **re**connect.

MohamedEbrahim13, 2024-09-28:
Dose the kable's support reconnection when the connection between the devices failed after the initial connection happen ?
I want to listen if the connection failed Retry to connect .
How can I handel that with kable's?

MohamedEbrahim13, 2024-10-02:
i want to thank you for your response it helps me a lot

eygraber, 2024-12-18:
The answer explaining `autoConnect` that is linked in the README states:

> If the connection was established through an auto connect, Android will automatically try to reconnect to the remote device when it gets disconnected until you manually call disconnect() or close()

If Kable is explicitly calling `disconnect` or `close` when the connection is lost, then the Kable documentation should call out that it will not try to reconnect if `autoConnect` is true.